Eomys/pyleecan #247: Run GUI error: no module named pyleecan.definitions

MVreemann, 2020-11-20:
Hey guys,

As of late i am having trouble loading the GUI from pyleecan. I recently moved pyleecan to another directory and eversince then the GUI just won't load. I am not quite sure what the problem is, since my git is up to date (i just updated it) i think i have all the right enviroment variables set and i do think all my packages are up to date.

The error message i got is:

`During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\menno\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\menno\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\menno\Documents\GitHub\pyleecan\pyleecan\__main__.py", line 2, in <module>
    from .run_GUI import run_GUI
  File "C:\Users\menno\Documents\GitHub\pyleecan\pyleecan\run_GUI.py", line 22, in <module>
    from definitions import PACKAGE_NAME, ROOT_DIR, config_dict
  File "C:\Users\menno\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'definitions'`

To me this seams like i have a problem in the shiboken2 package, but trying to update it with pip reveals that it should be up to date. I also updated PySide2 which seems to rely on shiboken2, but this also changed nothing.

Does anyone have an idea of what the problem might be?

All the best,

Menno

SebGue, 2020-11-20:
Hello Menno,

do you have PyQt5 and PySide2 installed? I also had problems due to that, so I uninstalled PyQt5, which solved my problem.

Best regards, Sebastian

MVreemann, 2020-11-20:
Hey Sebastian,

I just uninstalled PyQt5 like you said, but i still have the same error...
Besides that i ran the pytest, `python -m pytest -m "not long"` and it was interrupted after 49 errors, so i think something may be wrong with my install. although i'm not sure what.

All the best,

Menno

BonneelP, 2020-11-20:
Hello,

You said that you "moved" your pyleecan. Can you check your python path ? Maybe you have a version of pyleecan installed in your python and it is used instead of the local folder you are trying to use. 

Then in your error there is: 
File "C:\Users\menno\Documents\GitHub\pyleecan\pyleecan_main.py", line 2, in
from .run_GUI import run_GUI
File "C:\Users\menno\Documents\GitHub\pyleecan\pyleecan\run_GUI.py", line 22, in
from definitions import PACKAGE_NAME, ROOT_DIR, config_dict

We use relative import so maybe the script pyleecan_main introduce an import error. Usually we run directly the run_GUI script in its folder. Can you check that the import are correct within your script ? 

Best regards,
Pierre

MVreemann, 2020-11-20:
my python path is correct (i take it you mean in the system variables?) it refers to C:\Users\menno\Documents\GitHub\pyleecan. 

The weird thing is, i don't actually see a pyleecan_main.py file in my directory. eventhough the error sais it is used. So i don't know how to check the code

BonneelP, 2020-11-20:
If I understood well, you try to open the GUI with python -m pyleecan ? I think that this syntax is only for the official release installed with pip. Can you try running directly the run_GUI.py script ? When using "pip list", is there a pyleecan version listed ?

Best regards,
Pierre

MVreemann, 2020-11-20:
Specifically running run_GUI fixed the problem (although it turned out some packages were missing). I indeed had pyleecan 1.0.0 installed from pip. Would it be wise for me to uninstall this?

MVreemann, 2020-11-20:
alright, now the new problem is that somehow if i try to import run_GUI.

For a bit of context: I am trying to make a GUI from which you can easily start a simulation. This is useful for my colleges who can't write code, so probably are unable to initialize a simulation otherwise. 

As a start i am trying to import run_GUI into my script (in the GitHub folder for now.) to do this i call:

`from pyleecan.pyleecan import run_GUI`

Than i again get the same error that there is no module named 'pyleecan.definitions'

BonneelP, 2020-11-20:
Great :) 
When developing pyleecan, I recommand not to have it installed, otherwise you may encounter some issues. In particular you don't know what does it mean : import pyleecan ? The one from the script or the one from pip ?
I personally have a python for my IDE without pyleecan installed for development, and Anaconda (another python) with the latest pyleecan version for scripting and presentation. That way I always know which version of pyleecan I'm using.

If you want to create a new GUI, maybe you should copy/adapt run_GUI instead of importing it ;)

Best regards,
Pierre

BonneelP, 2021-02-05:
Hello,

Do you have any other question on this issue or can we close it ?

Best regards,
Pierre

MVreemann, 2021-02-05:
Yes you can close it!

closed 2021-02-05 by MVreemann