Eomys/pyleecan #249: [NF] Automated simulation GUI window

MVreemann, 2020-11-24:
Hello everyone,

I want to propose a new tab for the GUI. My idea is that in this tab you can easily call simulations with the press of a button. For example the simulation with FEMM as explained in the tutorials would be a good start. Since these types of simulations always require similar code to perform.

Now i have toyed around with creating a seperate GUI for this, but implementing it within the existing GUI would make more sense.

For this the window would need some general inputs in the form of bullit points, dropout menus or text fields. 

- The motor file location (which is loaded anyway in the GUI)
- The wanted simulation (dropdown: Magnetic, Electric, Structural, Force, Acoustic)
- Rotorspeed (text field)
- Stator current (text field) 
- Simulation time (text field)
- Angular and time discretization (text fields)

- Wanted output plots (Bullet points)

For the different modules you may need some other inputs, but that should not be a problem. If it works for the tutorial it should be fairly easy to expand to cover everything.

Now here's the thing, this shouldn't take too long, however i can't figure out the orginisation of your GUI code. It seems everything is initialized from .ui files, but i don't exactly how it works. Could someone perhaps explain me the general idea? Then i will move on to making the GUI add-on.

All the best,

Menno

BonneelP, 2020-11-24:
Hello,

We haven't add the time to write the how to contribute to the GUI tutorial. But the short version is:
- First you need to define a .ui file with Qt designer => This file define all the widgets and layout (composition/content of the GUI)
- Then we have gen_list.json which enables to create link between the csv doc and the GUI (to set tooltips, min/max) - this file is not mandatory (if none of the widget are listed in the csv)
- Finally there is the class file that defines all the signals and slot of the widget (what the GUI does when you click on X button)

The class generator also contains a GUI generator. For a widget like PWSlot10. The class generator convert PWSlot10.ui to Ui_PWSlot10.py that contains a Ui_PWSlot10 object. The class generator convert gen_list.json to Gen_PWSlot10.py that contains a Gen_PWSlot10 object that inherit from Ui_PWSlot10. Finally the PWSlot10.py is a file we have written that define a class PWSlot10 that inherit from Gen_PWSlot10 and that is used in the GUI. Take a look a this class, this is a basic example to understand the 3 file layers. 

Best regards,
Pierre

MVreemann, 2020-11-24:
Thanks Pierre! I'll take a look at it right away. This was precisely the information i was looking for.

MVreemann, 2021-02-15:
Alright, quite some time has passed now, but i finally have somewhat of a "working" widget. I only did some basic algorithms for SCIMs but i am running into a problem.

I made a script which works when i run it from the cmd prompt, but when i link it to a button in the gui it somehow doesn't work anymore. I always get an error from FEMM that it can't excess a file (pyleecan/results/EEC compparameters/FEMM/SCIM_010_example_machine_model_3) Sometimes there is a random other error.

I have tried and tried again, but somehow it never really works... If i call the function from another basic gui (one button in a window) it does work, so i think it must have something to do with the pyleecan GUI but i can't figure it out. Does anyone have an idea? I'll attach my files so far to this message.
[SCIMSimu files.zip](https://github.com/Eomys/pyleecan/files/5981537/SCIMSimu.files.zip)

Short explanation of how to get to my error:

-The SimuWidget folder should be placed in pyleecan/GUI/tools

-for now i have the SCIM folder in the pyleecan/pyleecan folder, which is messy but for now it's okay

-Run the run_GUI.py script and the widget should turn up.

-load a motor (i did SCIM_010). click the specific speed radio button. Plug in some random values for voltage frquency and speed and press start

-If all goes well FEMM should now start and after some point you hear a sound and a pop up says that there is some error

SebGue, 2021-02-15:
Hello Menno,

I think you missed to zip your version of run_GUI.
Nevertheless, would you mind to commit your changes to a dedicated feature branch in your repo. Otherwise there may be version issues, since everyone how download your zip may have an other commit state.

Best regards, Sebastian

MVreemann, 2021-02-16:
Alright, i updated my pyleecan directory to the current version and that seemed to have fixed the problem. Consider my question answered, even though i still don't know what went wrong