Eomys/pyleecan #253: [CO] build_geometry labels

SebGue, 2020-11-28:
Hello,

I think it would make some sense to introduce a default convention for the labels set in build_geometry. 
There should also be some convenience function to set and get the labels. So the actual label creation is removed from all build_geometry methods and will be in one 'central' place.

What do you think?

Best regards,
Sebastian

BonneelP, 2020-11-30:
Hello Sebastian,

Great idea ! We already talked a bit about it in #195. We need to take into account the constraints "Several laminations" from #212 and if possible the 30 characters constraint from #202.

I agree that having a method that gather all the labels would be handy. What about a method "get_label_dict" that would return a dict with key the surface type ("Winding", "Lamination", "Magnet", Vent"...) and the value would be the proper label "Stator_0_Lamination", "Wind_Stator_R0_T0_S0"... 
I think that this method can be defined in Lamination and always return the complete dict. That way all is defined in a single file.
Then maybe we can remove the "R0_T0_S0" part and let the build_geometry set it.
What do you think of this proposal ?

For this issue, I think that we should first set the label conventions by writing the proper documentation on pyleecan.org, then add this new method and finally update the labels in all the code (in particular FEMM).

Best regards,
Pierre

BonneelP, 2021-09-14:
Done in #412

closed 2021-09-14 by BonneelP