Eomys/pyleecan #255: [CC] Periodicity issue

SebGue, 2020-12-01:
Hello,

there seems to be an issue with the periodicity computation. I got a 20 pole / 18 slot machine that got a periodicity of 2, i.e. one periodicity is 10 poles / 9 slots.  Nevertheless `machine.comp_periodicity()` 'tells' that there is one periodicity with the a possible anti-perodicity.
Therefore e.g. the (automatic) winding is wrong (see fig. 1) and the simulation results are also invalid :-(

Fig. 1. Winding plot with wrong Phase A direction in slot 9 
[image]

Best regards, Sebastian

BTW: It's great that the automatic winding definition is able to setup such pole and slot numbers already (despite the periodicity issue).

EmileDvs, 2020-12-01:
Hello @SebGue,

First you can check that the winding plot is consistent with the winding matrix. Then I think the machine.comp_periodicity() method  (which finds two anti-periods) is consistent with the winding plot, because the anti-periodicity is respected.

Finally, has the unit-MMF the proper polarity (p=9) ? It should not be the case, because you expect a different winding pattern to have so. 

For me it is more a problem of algorithm to define winding pattern given Zs and p (see #235).

Best regards,
Emile

SebGue, 2020-12-01:
Hello @EmileDvs,

you are right, it is not an issue with periodicity but an issue with winding pattern algorithm. I setup a user defined winding and every thing work as expected now.
So I will rename this issue respectively. Maybe someone of you could have a look at the algorithm to find the issue or at least add some comments.

I will leave this issue open and assign it to me to add some unit tests to avoid such issues.
E.g. check sum of conductors as in WindingUD and test for expected mmfs of common slot/pole combinations.

Best regards, Sebastian

SebGue, 2021-01-31:
Hello to you all,

@EmileDvs: Can you tell me something about the tooth winding algorithm. Is there a source or is it Eomys own work?

Actually I am not too happy with the Windings at all. There seems to be a lot of flaws.
So I got some ideas, I would like to discuss.
- It may make some sense to have a method 'convert_to_WindingUD' to easily setup a user winding for editing.
- There are at least 2 resources/tools to get winding patterns ([emetor.com](emetor.com), koil). So it would make some sense to have proper import for these tools.
- Maybe it would also make sense to create a collection with common ready to use windings that are accessable in the GUI.

So what do you think?

Best regards, 
Sebastian

BonneelP, 2021-02-03:
Hello Sebastian,

I fully agree with your 3 points :) I hope to find some time to focus on this after the upcoming release. We also share some thought in #235

Best regards,
Pierre

EmileDvs, 2021-04-27:
Hello Sebastian, 

I hope you're doing well since last time. For your information, I have updated #235 with an additionnal development that should solve this issue. I'll keep you posted!

Best regards,
Emile

BonneelP, 2021-05-17:
Solved with #378

closed 2021-05-17 by BonneelP