Eomys/pyleecan #262: Discussions

SebGue, 2020-12-09:
Hello at all,

I just got a github notification about the new 'Discussions' (beta) feature. Do you mind to try that out?
If you don't maybe @BonneelP could enable it in the repo settings.

Best regards, Sebastian

BonneelP, 2020-12-11:
Hello Sebastian,

It looks interesting indeed :) I just activated this new tab.

Best regards,
Pierre

SebGue, 2020-12-11:
Perfect! But just tell me if I'm spam you with questions too much :-)

closed 2020-12-11 by SebGue

BonneelP, 2020-12-11:
No problem at all ;) We are always happy to receive notifications from the pyleecan community :) I only answer it today because Friday is now a dedicated pyleecan / research day for me. I will continue to work and answer on pyleecan the other days but I will try to focus on the most urgent issues.  

Best regards,
Pierre