Eomys/pyleecan #284: a bug of pip install pyleecan

lyhehehe, 2020-12-28:
Dear guys
    when I use "pip install pyleecan", I have found some bugs.
   If we use pip install pyleecan, it will not install the module "ezdxf", which is necessary for the pyleecan.
   Besides, the numpy 1.19.4 may not worked for the pyleecan, but numpy 1.19.3 can be worked.
   In addition, Pyside2 may have trouble when the python version is further than 3.8.

BonneelP, 2021-01-04:
Hello,

Thank you for reporting this :) 

Regarding ezdxf, I just checked and it is in the requirements so it should be installed. We will soon create a new release (once PR #283 will be merged), I will investigate this issue then. 
Regarding the issue with numpy it was corrected in the requirements but I don't think it is in the current release. It should be fixed with the new release.
Regarding PySide2, I didn't know thank you for noticing. I have updated the doc (https://pyleecan.org/get.pyleecan.html) to recommand a python version between 3.6 and 3.8. (I have also added the link to this page in the README as it was missing)

Best regards,
Pierre

BonneelP, 2021-01-05:
Hello,

My colleague Cedric found the trick. There was a mismatch between requirements.txt and setup.py which is used to generate the release. I have updated it so it should be fixed. I will check it after the next release.

Best regards,
Pierre

HamzaFarooq007, 2021-01-13:
Hello everyone,
Just to let you know. I also had a problem while launching the GUI of Pyleecan with latest versions. The error message stated that 'ezdxf' is missing. So I installed it using the below command, and it worked perfectly.

pip install ezdxf

Hamza FAROOQ

BonneelP, 2021-01-13:
Hello Hamza , 

Thank you for reporting this issue :) Normally it will be fixed with the next release (I hope this Friday). 

Best regards,
Pierre

BonneelP, 2021-02-05:
Hello all,

It was not the Friday I thought but we just released pyleecan 1.1.0 :) We took the time to investigate some bugs and add few extra features, sorry for the wait. 

I just installed it with pip on a fresh python installation and the GUI opens correctly and the proper version for numpy and ezdxf are installed. So everything seems fixed, thank you for reporting this issue. I'm closing it for now, feel free to reopen it if it needs further work.

Best regards,
Pierre

closed 2021-02-05 by BonneelP