Eomys/pyleecan #295: angle rotor

SebGue, 2021-01-12:
Hello to you,

actually it seem, that the input/output.elec property 'rotor_angle' is not not utilized e.g. by dq to uvw transformations of current in Input and Elec (where the frequency and time are utilized).
I think this should be fixed. Otherwise it will be possible, that the requested dq current may not be the actual dq current.

Further, there seems to be some inconsistency regarding the initial rotor angle. The value could be set in Magnetics but the actual initial rotor angle used for the simulation seems always be computed from machine.comp_angle_offset_initial() by Output.get_angle_offset_initial(). This is because output.geo.angle_offset_initial may always be None.

If some of you could give a hint, I will be very happy about it.
(This may also regard the last comments on #202)

Best regards, Sebastian

BonneelP, 2021-02-05:
Hello Sebastian,

Sorry for the long time without answer. I think the issue is linked to the fact that we tried several ways to implement this angle in pyleecan before finding comp_angle_offset_initial and that the old properties were not cleaned. So here some thought and propositions on how to reorganize things:
- The default behaviour should be to use comp_angle_offset_initial without any required parameters
- We should be able to enforce the initial position off all the laminations (even for >2). What about a list of initial position (in the lamination order) or a dict with lamination label as key (not in dict = 0 ?). Or maybe we can have a property for that in the lamination (but I'm not sure if we should mix machine and simulation parameters)
- The starting angle should be computed once, stored in Out.Geo. This computation need to be common for several models (elec, mag,...) and use the enforced initial positions. What about moving it to Input ? 
- Finally, we should be able to completely skip this computation to enforce the rotor position vector (which is done in Input)

Hope this will help you :) This part of the code is very important for all pyleecan simulation, so I agree that we should talk and set the corresponding architecture once and for all. 

Best regards,
Pierre

EmileDvs, 2021-12-09:
issue solved with PR #464

closed 2021-12-09 by EmileDvs