Eomys/pyleecan #307: `TypeError: __init__() got an unexpected keyword argument 'normalizations'` in pyleecan/Tutorials/tuto_Simulation_FEMM

MarkWengSTR, 2021-01-30:
Hi, 

Thanks for building this amazing project,

The issue occurs while I try to running the jupyter-notebook tutorial (tuto_Simulation_FEMM),
it was work until run
`rot_dir = simu_femm.machine.stator.comp_rot_dir()`

it will got an error 
```
TypeError                                 Traceback (most recent call last)
<ipython-input-4-230e2e5aa176> in <module>
     28 I0_rms = 250/sqrt(2)
     29 felec = p * simu_femm.input.N0 /60 # [Hz]
---> 30 rot_dir = simu_femm.machine.stator.comp_rot_dir()

c:\python37\lib\site-packages\pyleecan\Methods\Machine\LamSlotWind\comp_rot_dir.py in comp_rot_dir(self)
     20     # Compute unit mmf
     21     MMF, _ = self.comp_mmf_unit(
---> 22         Nt=20 * p, Na=20 * p
     23     )  # 20 points per pole over time and space is enough to capture rotating direction of fundamental mmf
     24 

c:\python37\lib\site-packages\pyleecan\Methods\Machine\LamSlotWind\comp_mmf_unit.py in comp_mmf_unit(self, Na, Nt, freq)
     59         symmetries={"period": per_a},
     60         values=angle,
---> 61         normalizations={"space_order": self.get_pole_pair_number()},
     62     )
     63     Phase = Data1D(

TypeError: __init__() got an unexpected keyword argument 'normalizations'
```
How can I fix it?

thanks for your help.

SebGue, 2021-01-30:
Hello Mark, 
I guess you have to upgrade your SciDataTool.
Best regards, Sebastian

MarkWengSTR, 2021-01-30:
Hi  Sebastian,
it's work after I upgrade SciDataTool package.
thanks for your help

closed 2021-01-30 by MarkWengSTR