Eomys/pyleecan #308: [GUI] Error when try to run "run.GUI.py"

IvanMikulic1997, 2021-01-30:
Hello all,

I recently installed Pyleecan and I am having trouble with running the "run.GUI.py" script.

When I run it I got the following errors:

_Traceback (most recent call last):
  File "C:/Users/Ivan Mikulić/AppData/Local/Programs/Python/Python38/Lib/site-packages/pyleecan/run_GUI.py", line 6, in <module>
    from PySide2.QtCore import QTranslator
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\Lib\site-packages\pyleecan\PySide2\__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\Lib\site-packages\pyleecan\PySide2\__init__.py", line 55, in _setupQtDirectories
    os.add_dll_directory(dir)
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\lib\os.py", line 1109, in add_dll_directory
    cookie = nt._add_dll_directory(path)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\Ivan Mikulić\\AppData\\Local\\Programs\\Python\\Python38\\Lib\\site-packages\\pyleecan\\libpyside'_

I would really appreciate your help on this because I don't know where is the issue.

SebGue, 2021-01-30:
Hello Ivan,
it seems to be some issue with PySide. Are you able to import PySide manually?
Best regards, Sebastian

IvanMikulic1997, 2021-01-30:
Hello Sebastian,

How do you mean "import PySide manually"? 

KR, Ivan

SebGue, 2021-01-30:
Just open a python console and do 'from PySide2.QtCore import QTranslator'.
Or you create a python script with this import and run it, e.g. with idle.

IvanMikulic1997, 2021-01-30:
Ok, I got it.
I put these two lines of code

_from PySide2.QtCore import QTranslator
translator = QTranslator()_

And I got the following error:

_Traceback (most recent call last):
  File "C:/Users/Ivan Mikulić/AppData/Local/Programs/Python/Python38/Lib/site-packages/pyleecan/proba.py", line 1, in <module>
    from PySide2.QtCore import QTranslator
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\Lib\site-packages\pyleecan\PySide2\__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\Lib\site-packages\pyleecan\PySide2\__init__.py", line 55, in _setupQtDirectories
    os.add_dll_directory(dir)
  File "C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\lib\os.py", line 1109, in add_dll_directory
    cookie = nt._add_dll_directory(path)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:\\Users\\Ivan Mikulić\\AppData\\Local\\Programs\\Python\\Python38\\Lib\\site-packages\\pyleecan\\libpyside'

Process finished with exit code 1_

SebGue, 2021-01-30:
What is your output if you move your script to another dir other than pyleecan, e.g. to C:\test

SebGue, 2021-01-30:
Could you also run the following code

```
import sys >>> sys.path
```

SebGue, 2021-01-30:
... you should also make sure that python is in your enviroment variable 'path', i.e. 'C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\' and 'C:\Users\Ivan Mikulić\AppData\Local\Programs\Python\Python38\Scripts\' in your case.

SebGue, 2021-01-30:
[image]

IvanMikulic1997, 2021-01-30:
I checked the Path and everything is there as on the picture that you sent.

IvanMikulic1997, 2021-01-30:
[image]

IvanMikulic1997, 2021-01-30:
I have run this code in the C\test and it seems that it worked

IvanMikulic1997, 2021-01-30:
But now I don't get it why it don't work in the previous case?

SebGue, 2021-01-30:
I don't get it eigther. But if your script is working in C/test you could try to also run pyleecan there, i.e.
```
from pyleecan import run_GUI

run_GUI()
```
Probably it is not possible to run a package from within its installation dir.
But I can not rreproduce this since I have not installed pyleecan with pip.

IvanMikulic1997, 2021-01-30:
Thanks for the help. I successfully opened the GUI.

closed 2021-01-30 by IvanMikulic1997