Eomys/pyleecan #309: Torque wave not same value as tutorials example in tuto_Simulation_FEMM.ipynb

MarkWengSTR, 2021-01-30:
Hi

I download the tuto_Simulation_FEMM.ipynb and just run every cell step as setup.
the radial and tangential air gap waveform are same as the example.
[image]
[image]

but the torque wave is **not** like the tutorial's example
[image]

correct torque waveform in your example
[image]

seems like same speed or excitationsetting problem.

could you help me to fix this problem?

thanks

SebGue, 2021-01-30:
Hello Mark,
I can't try this at the moment myself. But you could first plot Phi_wind_stator to see whats going on, i.e. the number of revs. From the curve it seems that the exitation is okay. 
Then you could set time periodicity to False as an intermediate solution and see if this works.
Best regards, Sebastian

Superomeg4, 2021-01-31:
Hello everyone,

This issue may be related to the issue #301 (I'm working on it next week). 

@MarkWengSTR Did you run the tutorial with the Pyleecan version installed from pip (last release) ?

MarkWengSTR, 2021-01-31:
ohoh, Yes, I run it with the last release version.

BonneelP, 2021-01-31:
Hello, 

Which version of SciDataTool are you using ? Can you try to run this simulation without symmetries to see what is the result ?
I agree with @Superomeg4, this may be linked to #301. We are currently investigating the validations cases of pyleecan. Once all the tests will be Ok, #304 merged and #301 solved we will make a new release (and update the tutorials afterwards). I will focus on this objective for now. 

Best regards,
Pierre

MarkWengSTR, 2021-02-01:
I use SciDataTool  1.2.2 version，the torque waveform is correct without symmetries，thanks
[image]

Superomeg4, 2021-02-02:
Hello @MarkWengSTR,

I tried to relaunch the tutorial with the master version of Pyleecan, and the last version of SciDataTool, and I get the "correct" torque waveform. 

If you don't want to wait until the new release, you can go into your local folder of the pyleecan code and use 
`pip install -e .`

Best regards,

Raphaël

BonneelP, 2021-02-05:
Hello, 

We just released pyleecan 1.1.0, you can install it with pip install -U pyleecan. 
Is everything fixed on your side ? Can we close this issue ? 

Best regards,
Pierre

MarkWengSTR, 2021-02-07:
It's fixed.

thanks very much

closed 2021-02-07 by MarkWengSTR