Eomys/pyleecan #311: The plot closes imediately after it opens

IvanMikulic1997, 2021-02-03:
Hello,

I believe that this is a simple problem, but I am not so good at programming, so here it is.
I am going through the first webinar example and in Jupyter notebook everything works fine, but when I try to implement the same example in PyCharm my plot closes immediately as it opens. The plot commands are following:

_im=IPMSM_Prius_2004.plot()
out_femm.plot_2D_Data(...)_

Also, when I try to hold the plot on the screen with _input()_ it seems that the python isn't drawing anything on that plot, i.e. it freezes.

Kind regards,
Ivan

BonneelP, 2021-02-03:
Hello Ivan,

In your script can you try to add "plt.show()" after calling you plot command ? (with import matplotlib.pyplot as plt)
Pyleecan have changed since the first webinar, if you have any strange results feel free to contact us ;) 

We are about to create a new release, I will see if I can take the time to update the tutorials afterwards. 

Best regards,
Pierre

IvanMikulic1997, 2021-02-03:
Hello Pierre,

This fixed the problem with the showing of plots, but now the window stays open and the code is paused until I manually close the window.
Is there a way in python to show multiple windows?

Kind regards,
Ivan

SebGue, 2021-02-03:
Hello Ivan,

you could try 'plt.show(block=False)'.

Best regards, Sebastian

IvanMikulic1997, 2021-02-03:
Great, this solved the problem.

Thank you all for the help.

closed 2021-02-03 by IvanMikulic1997