Eomys/pyleecan #317: wrong units in the function create_FEMM_circuit_material

harshasunder-1, 2021-02-13:
Hello,
I noticed that in the function create_FEMM_circuit_material the variable 'Swire' from lam.winding.conductor.comp_surface() is in m**2. The diameter is then calculated in mtrs and used in set_FEMM_wind_material(). However that function requires it to be in mm, as seen in the comments as the top of that function. It is used within the function in mi_addmaterial() which requires it to be in mm as per the FEMM manual.
Best Regards
Harsha

SebGue, 2021-02-14:
Hello Harsha,

thank you for noticing. I just fixed it.
Do you run harmonic analysis that were affected?

Best regards,
Sebastian

harshasunder-1, 2021-02-16:
Hello Sebastian,

Thanks.
Actually I noticed the diameter wrong in the FEMM material properties window when I was just checking the material properties of a winding in one of my projects. Didnt actually affect me in that work :)

Best Regards
harsha

closed 2021-02-16 by harshasunder-1