Eomys/pyleecan #319: [NF] Multisimulation of a Multisimulation

SebGue, 2021-02-14:
Hello to all,

this issue is to discuss a new feature "Multisimulation of a Multisimulation". This feature could be usefull e.g. if one wants to do a parameter sweep on some machine parameters while calculating multiple operational points.

There also was a discussion (#302) on that feature before. I also started to do an implementation so it may be less abstract to discuss some details (see #318).

I first thought it would be a good idea to have a 'ref_simu' property on VarSimu to store a reference simulation that could also be a multisimulation (i.e. has a VarSimu contained). But I skipped that idea because it would require some additional logic to define which is the actual reference if there is 'simu' and 'simu.var_simu.ref_simu'. This may complicate things.
So I came up (thanks to @BonneelPs answer about layers) with the idea to add a 'var_simu' property to VarSimu to realize layers of multisimulations.

Going through the code, fortunately there is a line of code where 'var_simu' of the multiple simulations (that are copied and modfied reference simlations) is set to None to avoid infinite loops. There I just set the next layer to 'var_simu'. Also I had to 'allow' XOutput as element of output_list.
The maybe most important change is in 'copy' and 'as_dict'. Setting the (one) next layer to multiple 'VarSimu.var_simu' requires to copy the respective VarSimu. Otherwise there will be issues regarding 'parent' property.
Since 'copy' and 'as_dict' will discard all functions (that are not of type string) and copies of VarSimu would not be working any more, I had to change that as well. So I added kwargs to optionally keep functions.

I ran a (parallel) test on it and it seems to work quiet well. 

So, I hope you like and agree with that implemention. But if you see any short comings or serious issues, I'm open to do it any other way.

If we stick to it there is still some things to do. Most important, there have to be custom setters to avoid infinite loops. Further there also is the idea of preprocessing property to set all (or at least some) parameters at once, that has not been implemented yet. This also requires to intermediate store the parameters until they are set (ideas welcome).
And finally there need to be some good unit tests.

So far...

Best regards, 
Sebastian

BonneelP, 2021-02-16:
Hello Sebastian,

I haven't the time yet to go in depth into your PR but I would like to answer some aspects of your issue. 

Regarding "infinite loop" at first we were thinking of adding a new property for simulation to contain this second layer of multisimulation (var_simu_2 for instance). The idea would have been to "just" (I haven't confronted this idea to the reality of the code) modify the simulation.run method to check first var_simu2, if defined we call var_simu2.run that would create a list of simulation with var_simu2 = None and var_simu != None. Then when calling run on these simulations check var_simu2 is False, check var_simu would create a list of simulation with var_simu2=var_simu=None. The point is to gain in abstraction, a var_simu creates a list of simulations (whatever it also contains a var_simu). 

Regarding functions normally the new standard is to always use string but it is interesting to see that you find a way to copy the function for this case and not for save/load :) 

If I understood your implementation, to create a parameter sweep of VarLoadCurrent , then type(simu.var_simu) is VarParam and type(simu.var_simu.var_simu) is VarLoadCurrent ? 

> If we stick to it there is still some things to do. Most important, there have to be custom setters to avoid infinite loops. Further there also is the idea of preprocessing property to set all (or [...]

Can you further explain this part ? 

Best regards,
Pierre

SebGue, 2021-02-17:
Hello Pierre,

> If I understood your implementation, to create a parameter sweep of VarLoadCurrent , then type(simu.var_simu) is VarParam and type(simu.var_simu.var_simu) is VarLoadCurrent ?

Yes, that's exactly way one should do that, e.g. to do some postprocessing on the whole set of operation points. But one can also have the parameter sweep as the 'second' layer and the operations points as the 'first'.
One could also favor an 'one layer per parameter' strategy if that make sense for some reason. It is totally up to the users.

> Can you further explain this part ?

I just wanted to say that, If you like this concept/implementation the next step would be to care for the infinite loops.
The best way, I think, is to do it with the setter of var_simu property. Otherwise there also will be infinite loops e.g. in as_dict method.

A further idea (regardless of the infinite loops) is to have a preprocessing property similar to OptiProblem. Then it would be possible to set some (dependent) machine parameters at once. But this requires to get access to the actual simulation parameters that should be set. But I didn't had an good idea yet.

Best regards, Sebastian

BonneelP, 2021-02-17:
Hello Sebastian,

> Yes, that's exactly way one should do that, e.g. to do some postprocessing on the whole set of operation points. But one can also have the parameter sweep as the 'second' layer and the operations [...]

Great :) We can also do as many layer as we want with this idea.

> I just wanted to say that, If you like this concept/implementation the next step would be to care for the infinite loops.

If I understood well your implementation the behaviour is as follow:
- simu.run check for var_simu
- in var_simu.run when generating the simu list, instead of setting simu.var_simu = None, we set simu.var_simu = simu.var_simu.var_simu. (pyleecan/Methods/Simulation/VarLoadCurrent/get_simulations.py & pyleecan/Methods/Simulation/VarParam/get_simulations.py)
Then the only thing missing to avoid infinite loop is to handle the reference simulation with the same logic both if the simulation is in the multisimulation (VarSimu.run, which is already done since the simulation is generated with get_simulations) and when it is not the case (simulation.run line 36, we need to copy the ref simulation and set simu.var_simu = simu.var_simu.var_simu before calling run instead of run_single).
Note shouldn't we be calling run instead of run_single for the generated simu list as well ? The difference between run and run_single is the check on var_simu and now the generated simu can have var_simu != None as well.
I don't see what need to be changed in the setter of var_simu property to avoid infinite loop but maybe I'm missing something obvious once you start to run this kind of simulation :/ as_dict and copy skip the parent part if I remember well. 

> A further idea (regardless of the infinite loops) is to have a preprocessing property similar to OptiProblem. Then it would be possible to set some (dependent) machine parameters at once. But this [...]

We can add a preprocessing the same way we did for the properties "pre_keeper_postproc_list" and "post_keeper_postproc_list" but then it should be separated from the variable simulation definition. As ParamExplorer already enable to use scripts we already should have all the flexibility to generate what we want. For example there is Tests\Validation\Simulation\Multisimulation\test_slot_scale.py that already use a script to scale a machine. What do you plan to use the preprocessing for ? 

We also want to introduce the Multisimulation of a Multisimulation feature in the near future. Do you mind if I join your work directly on your branch ? We can also schedule a phone call if you want. 

Best regards,
Pierre

SebGue, 2021-02-18:
Hello Pierre,

you are right on the implementation behaviour. For the use of run vs. run_single you proposed, I wanted to change as little code as possible. But if you seen some benefits, I don't mind.

The infinit loop thing seems clearly to be some user issue as far as I could see. Nevertheless we should address it. 
My initial proposal was to do it in the var_simu setter instead in VarSimu.run. 
The reason was that a loop will also occur on the as_dict method. But maybe that's the solution ... (Just figured out while writing this :-) )
```
from pyleecan.Classes.VarParam import VarParam
var_simu = VarParam()
var_simu.var_simu = var_simu
```
It would somehow be stupid to do that, nevertheless it could happen in a more complex (multilayer) simulation.

Now, the mentioned solution could be in the run method still. To check a simulation we'd first try to do `self.as_dict()` on a simulation. If there is a loop, python seems to have a mechanism, that will raise an error if there is a max. recursion depth exceeded. If that is the case the simulation shouldn't be started.

I think I will not work on the branch uptill next week. So you can work on it directly without issues, if you like. But you can also give me a call whenever you want.

The preprocessing is maybe worth an other branch/issue. The slot scale test you mentioned has only one parameter.
I had a multisimulation where I had to set multiple (non-linear) dependend parameters. I.e. I wanted to set the magnet height (and therefore also the rotor diameter) and the magnet width (in given as pole coverage). Now the 'problem' was to maintain the airgap to fit a rotor sleeve with its thickness depends on the rotor diameter as well as the magnet size.
This problem is basically easy to solve by some iteration technique. Only it requires to have all the parameters availiable while with the actual implementation there is a one setter per parameter approach.
So I hope I could point out my issue with this example. Maybe you got some other idea on it.

Best regards,
Sebastian

BonneelP, 2021-02-19:
Hello Sebastian,

I take the time to rewrite the complete process to make sure that everything is clear. Here is my new complete proposal:

First we have the "normal" simulation: When calling simulation.run, var_simu is None, then run_single is called which call run for each module and an Output is returned

Then there is the multi simulation: 
1) When calling simulation.run, var_simu is not None so var_simu.run is called. To simplify the process Simulation.run, will now only check var_simu == None, check Output type, init the logger and call either run_single or var_simu.run => the call to run_single of the reference simulation will go into var_simu.
2) We create the reference simulation with ref_simu=simu.copy(keep_function=True) and ref_simu.var_simu = simu.var_simu.var_simu which is None (in this case)
3) the list of simulation is created by calling get_simulations. To create these simulations we only need the reference simulation and the modification rules. 
4) Then we run the reference simulation (that can be either ref_simu, or a simulation from the simulation list). This is why get_simulations is called before. When calling ref_simu.run, run_single is called since var_simu is None.
5) It returns an Output object (ref_out). We use this output object to introduce some shortcut in the simulation list with set_reuse_data (for now mostly to reuse the FEMM file). 
6) Then we can run each simulation from the list. Each simu have var_simu == None so run_single is called. At the end of the run we call the pre_keeper_postproc_list, then the DataKeeper, then post_keeper_postproc_list on the Output object. Each run will return an Output object that can be added the output_list.
7) At the end we create an XOutput object, the Output part is extracted from the Output of the reference simulation (ref_out), the XOutput part is the DataKeeper and the output_list.

So now for the multi simulation of multi simulation.
1) When calling simulation.run, var_simu is not None so var_simu.run is called. 
2) We create the reference simulation with ref_simu=simu.copy(keep_function=True) and ref_simu.var_simu = simu.var_simu.var_simu **which is not None**
3) the first layer list of simulation is created by calling get_simulations. To create these simulations we only need the reference simulation and the first layer modification rules. 
4) Then we run the first layer reference simulation (that can be either ref_simu, or a simulation from the simulation list). This is why get_simulations is called before. 
    4.1) When calling ref_simu.run, the second var_simu.run is called since it is now not None
    4.2) We create the second layer reference simulation with ref_simu=simu.copy(keep_function=True) and ref_simu.var_simu = simu.var_simu.var_simu **which is None** (in this case but we can add other layers, at the end the default value of var_simu is None so it will eventually ends)
    4.3) the second layer list of simulation is created by calling get_simulations. To create these simulations we only need the second layer reference simulation and the second layer modification rules. 
    4.4) Then we run the 2nd layer reference simulation (that can be either ref_simu, or a simulation from the simulation list). 
    4.5) Since the 2nd layer has var_simu == None, an Output object is returned. We use this output object to introduce some shortcut in the 2nd layer simulation list with set_reuse_data (for now mostly to reuse the FEMM file). 
    4.6) Then we run each simulation from the list of the reference simulation. Each simu have var_simu == None so run_single is called. At the end of the run we call the 2nd layer pre_keeper_postproc_list, then the DataKeeper, then post_keeper_postproc_list on an Output object. Each call to run will return an Output object that can be added the output_list.

5) The call the ref_simu.run of the first layer returns an XOutput object. We use this XOutput object to introduce some shortcut in the first layer simulation list with set_reuse_data (for now mostly to reuse the FEMM file). 
6) Then we can run each simulation from the first layer list. Each simu have var_simu != None so we run a multi simulation for each with the same principal as the reference simulation (cf 4). At the end of the run we call the pre_keeper_postproc_list, then the DataKeeper, then post_keeper_postproc_list on an XOutput object. Each call to run will return an XOutput object that can be added the output_list.
7) At the end the result is an XOutput object, the Output part is the Output part of the XOutput of the first layer reference simulation (which is in fact the Output of the second layer reference), the XOutput part is the DataKeeper and the output_list.

With all that said few hints on how to make it works:
- I think that the second layer must have is_keep_all_output = False, otherwise the memory will burn
- Then we need to set the DataKeeper of the first layer to use data from the DataKeeper of the second layer since we don't have the second layer output list. 
- Exemple: We run an Sensitivity study on a variable speed simulation. We can use the second layer DataKeeper to get the average torque for all speed. Then we use the first layer to get the effect of the input parameter on the torque curve as a function of speed or on the maximum torque for instance.

This afternoon I will start to experiment on this proposal. Feedbacks are very welcomed.
Best regards,
Pierre

SebGue, 2021-02-19:
Hello Pierre,

I hope I understood your proposal well.
I think the main difference between your proposal and my PR is the reference simulation and in the way of setting the output.
If your proposal works out, I'm fine. 
I only have one need. The multisimlation must be able to work with inherited XOutput class (maybe also inherited Output if that make sense), since I got some inherited 'special purpose' Simu1 and XOutput (that are mainly to distinguish result types). But that may not be a problem if you do something like `xoutput = type(simulation.parent)()` to get the output.

On the hints: I would not forbit is_keep_all_output since the results may be needed. Instead the user should care, e.g. with some post (datakeeper) processing to delete or compress some data.

I know this is not too much feedback, but I think next week I got some more time.

Best regards,
Sebastian

BonneelP, 2021-02-19:
Hello Sebastian,

Thank you for your answer. 
I'm updating the structure of the program to always use "xoutput = simu.run" instead of using the "predefined" Xoutput object. So it will be compatible with other output/simulation objects ;) I also take the opportunity to improve the readability of the main methods. 
I won't forbid is_keep_all_output, I just don't recommand it (but it depend of the simulation you are defining). The point is to handle any layer as if it was the first one. So all options should available be for all the layers. 

Best regards,
Pierre

BonneelP, 2021-02-19:
Hello all,

I just pushed my modifications for the multi-simulation of multi-simulation and it seems to work :) All the work is mostly on the VarSimu.run method which has been reorganized (some files were rename to better understand the global process). At the end one of the main difference is how we handle the xoutput and the reference simulation. Before we always use parent, now as there can be several XOutput, run_multisim_step returns the xoutput of the "simulation step". 

To go further, I could be interesting to improve the print/log to better follow what is going on (maybe by introducing a "layer" property in var_simu as we did with "index" in simulation.
The test_multi_multi is not really interesting (it is more a proof of concept), I will update it to make something more understandable and more valuable. 
I also need to further check how all the tests are behaving but the one I run were Ok.

Best regards,
Pierre

SebGue, 2021-02-23:
Hello Pierre,

I started (sanity) testing your latest code (while I haven't actually looked at the implementation details).
I think I got some misunderstanding what 'ref_simu_index' really means. I thought that the reference simulation will simply be added to the list of outputs. But that seems not to be the case.
In other words, has the reference simulation to be part of the parameter space, if ref_simu_index != None? And do I have to care which index of the parameter space the reference machine need to have?

I.e. I have some simple 2 layer simulation where I vary magnet height (layer 1) and current (layer 2). The reference machines magnet height is outside the parameter space. While out.simu.machine is the reference machine, the results (e.g. out.mag) are equal to a 'variated' machine.

Maybe some check is needed here for such cases.

Best regards, Sebastian

SebGue, 2021-02-23:
... here is another issue, I think.

My parameter space is 2 by 2, i.e. 2 parameter per layer. Now if I set layer 1 'ref_simu_index' to None, I would expect 1 + 4 = 5 simulations to run but it actually runs 6 simulations. I guess, it is because the reference simulations also run as a multisimulation while it's XOutput results are discarded. So there is some unnecessary calulation overhead.

BR, S

BonneelP, 2021-02-23:
Hello Sebastian,

This is a very important point indeed. The reference simulation is only meant to provide some data to setup/speed-up the other simulations. It also provides some default plot (if ones calls xoutput.plot_2D_Data it will use the output of the reference simulation). This architecture is interesting because running the reference simulation enable to check that everything looks right and enable some speed improvements. But with that, you always need to run a reference simulation before running all the others. So when defining a parameter sweep with N simulations, in fact you run N + 1 simulations. 

But most of the time, we don't care about the reference simulation. That is why we introduced the "ref_simu_index" to "skip" the run of the reference simulation. Then you pick the index of the simulation from the list you want to use as the reference (for instance because it is the simulation with the scalling factor=1) and it will be used as reference and added to the datakeeper (which is not the case if ref_simu_index==None). Then you run only N simulations instead of N-1. 

So most of the time, if the reference simulation doesn't matter, you can just pick ref_simu_index=0 to speed-up the computations (in particular if is_keep_all_output==False since output are discarded). 

Regarding why you are having 6 simulations I expect one of the following:
- var_simu.run => Parameter sweep, ref_simu_index!= None => runs 2 times var_simu.var_simu
- var_simu.var_simu.run => VarLoadCurrent, ref_simu_index==None => runs a reference simulation then 2 simulations
or 
- var_simu.run => Parameter sweep, ref_simu_index== None => runs a reference simulation and 2 simulations from the list
- var_simu.var_simu.run => VarLoadCurrent, ref_simu_index!=None => run 2 simulations
For the second case, the trick is that the reference simulation of the Parameter Sweep is also a variable simulation that runs would 2 simulations.
Then in both case 2*3 = 3*2 = 6

To get only 4 simulations, once needs to set ref_simu_index=0 for both VarSimu. So I would say that I behave as it should but indeed it can be confusing. 

Best regards,
Pierre

BonneelP, 2021-02-23:
Hello all,

On my side I'm done with the Multisimulation of Multisimulation. I took the time to work on the logging to make sure that it runs as it should and here is the results:
[image]
There is a new option "layer_log_warn" to filter the log starting from a particular layer. I have also added a "layer" property to simulation as well as a "NAME" constant for the VarSimu daughters (for logging). 
"test_multi_multi" is now more meaningful by following the MTPA curve of the Prius. Both layer have ref_simu_index=0. I will come back later on this test for communication purposes. 
[image]

All "not long" tests are Ok, I still need to run the long ones. 

Best regards,
Pierre

BonneelP, 2021-02-24:
Hello all,

I run all the Tests and after few correction the only one failing is:
[image]
But I'm not sure that it is linked to the this PR. Maybe #324 and #327 but I don't think that it should have change this test either.

Best regards,
Pierre

SebGue, 2021-02-24:
Hello Pierre,

since this also occur on master (after fixing 'is_get_meshsolution') it is not related to this issue. I will care about that later.

> To get only 4 simulations, one needs to set ref_simu_index=0 for both VarSimu. So I would say that it behave as it should but indeed it can be confusing.

I think we should set the var_simu prop. of the reference simulation to None if the ref. simu. if refsimu_index == None. 
Otherwise there will be (depending on the number of parameters) very much calculations and results that will be discarded.
One may want to have a reference machine that is outside parameter space, so forcing ref_simu_index != None is not an option in my opinion.

Further I think in case of ref_simu_index != None, output.simu should be set to the reference simulation. Otherwise it will be a mismatch between the simulation output and the simulation. A warning on overriding the simulation should be fair enough.

What do you think?

Best regards,
Sebastian

SebGue, 2021-02-24:
Hello Pierre,

this issue seems to be really challenging indeed and there are a lot of things to consider. Do you think you may finish this before your holidays? Otherwise I would propose to do some visualization meanwhile (e.g. a flow chart or something similar) to be clear on the different possiblities and all the details.

Best regards, Sebastian

SebGue, 2021-02-25:
Hello Pierre, 

I just realized your holiday only starts tomorrow evening, so you may indeed finish this before.

First let me come back to my idea (my second last post) to have ref. simu. with ref_simu_index = None and ref.simu.var_simu=None. I thought this could be useful in following case:
One has a reference machine with a given torque at a certain current (that's the ref_simu). Now you would like to compare the results of a machine parameter sweep (e.g. on material or some dimensions). Since you don't know the current you need to achive the ref. torque, you will also do a parameter sweep on range of currents (as the 2nd layer). So you will have to do 1 + N*M calculations (M is for the number of currents, N for the design parameter). 
Now if the reference simulation will also do a parameter sweep you got M - 1 unnecessary calculations.
So that's why I did my suggestion. To be honest, I don't know if I will ever use this and if I will need it, I could also split it into 2 simulations.

So consider this done. Next will be your suggestions ...

SebGue, 2021-02-25:
...

> We always run the reference simulation, which is defined as it is currently (simu.var_simu = simu.var_simu.var_simu) to be able to have "variable datakeeper".
Yes
>We call the DataKeeper on the reference simulation (maybe store it in datakeeper.ref_result with no type check?)
    After generating the simulation list but before setting the "set_reused_data" we check if a simulation matches the reference one.
> If yes we copy the result of the reference one and skip its run, otherwise we run all the simulations.
I think a 'datakeeper.ref_result' and coping results would not be a good idea. Instead, if the reference simulation is 'out of the grid', its results could be stored as the last index of output_list. So the var_simu results will always be index 0 ... N.

> We add a warning in VarLoadCurrent if no operating point from the OP_matrix matches the reference one (because it is a shame to run a "speed for nothing")
The same here as above.

> We remove the ref_simu_index property (no more headache :D)
Nevertheless, there should be a output property or a getter to know the ref_simu index.

> With that the "reference output" will always match simu.var_simu=None so we can access the corresponding plot and we have access to the reference DataKeeper data for the plot. Then there is no need to overwrite the simulation in output which I prefer not to.
Yes I know, altering user input data is not a good idea. :-)

All in all, the aim always should be a clear, easy to understand solution/workflow. I think that's the case with your proposal.

Best regards, Sebastian

BonneelP, 2021-02-25:
Hello Sebastian,

> I think a 'datakeeper.ref_result' and coping results would not be a good idea. Instead, if the reference simulation is 'out of the grid', its results could be stored as the last index of output_list. So the var_simu results will always be index 0 ... N.

This was my first thought, but I think it can be confusing to add the reference simulation in the result list. In particular if one try to get the number of simulation by checking len(result). In my commit from yesterday evening, I introduced a "result_ref" property with no type (with some class generator black magic) to separate it from the list.

> Nevertheless, there should be a output property or a getter to know the ref_simu index.

It won't be hard to add that, for now there is a log message to indicate that we skip a computation. Why do you think we would need to remember that a computation was skipped (since the results are complete ?)

Best regards,
Pierre

SebGue, 2021-02-25:
Hello Pierre,

I think it is not confusing to add the ref. simulation results to the result list since the user should know if it is part of the parameter space.
So if the ref. simu. is part of the var_simu, do you still copy the results to result_ref? That would at least require additional memory.

> It won't be hard to add that, for now there is a log message to indicate that we skip a computation. Why do you think we would need to remember that a computation was skipped (since the results are complete ?)

I don't understand what you mean here. I just thought (in case ref. simu. is in the var_simu, i.e. part of the parameter space) it would be handy to easily get it's result list index.

Best regards, Sebastian

BonneelP, 2021-02-25:
Hello,

I just pushed my latest modifications with some bug correction and log improvement here is the result:
[image]

> So if the ref. simu. is part of the var_simu, do you still copy the results to result_ref? That would at least require additional memory.

result_ref is a field of DataKeeper so it contains a matrix, float, int... that the user asked to extract and it shouldn't be too heavy (it is not the xoutput). The "copy" process is on line 134 of VarSimu.run and it should use pointer. I want to split reference and simulation list and copy the data for the consistency of the simulation/result list. There is always a result_ref, result list are always the correct length (which is the number of requested simulations) that matches the paramexplorer. I prefer not to add the ref result to the list because there is nothing to indicate that this value is particular. result_ref is more clear for me and it we be easy to understand in the plot script what we are doing (instead of getting result[-1])

> I don't understand what you mean here. I just thought (in case ref. simu. is in the var_simu, i.e. part of the parameter space) it would be handy to easily get it's result list index.

For me ref_simu_index is now a fully internal parameter (define line 39 of VarSimu.run and used line 114 of VarSimu.run). With the consistency constraint explain above, the user doesn't need to know that we skipped a step the result structure and meaning is always the same. There is a log to indicate that we skipped a computation (cf image above).

Best regards,
Pierre

SebGue, 2021-02-25:
I think there is some missunderstanding here. I was mainly talking about the output_list. 
So the for me is, where do you store the XOutput of the reference simulation?

BonneelP, 2021-02-25:
Oh you are right, sorry for the misunderstanding. For now we just don't store the reference XOutput except if it is in the simu list. We only store the output part of the reference simulation. Do you think that we should finfd a way to store it as well? Then I agree ref simu index can be handy (except if we use pointer)

SebGue, 2021-02-25:
Yes we should find a way to store xoutput, e.g. with a prop. 'xoutput_ref' (but only populate it if ref. simu. is outside the parameter space).
Pointer won't work with load/save or at least will waste memory and/or disk space.

BonneelP, 2021-02-25:
Ok then we need to do the following:
- Add a new property xoutput_ref to Xoutput
- Add a new property xoutput_ref_index to XOutput
- When running the varsimu, only if is_keep_all_output is True, we either set xoutput_ref or xoutput_ref_index depending on the value of ref_simu_index.

Thank you for spotting this. I will add it.
Regards,
Pierre

BonneelP, 2021-02-25:
I will also add a get_xoutput_ref method to handle this double definition. Note that we duplicate the output part of the reference simulation but only when is_keep_all_output is True which already imply a large output so it should be OK.

closed 2021-03-09 by BonneelP