Eomys/pyleecan #321: [BC] save h5

SebGue, 2021-02-17:
Hello,

it's me again :-) I think I found another bug in save h5, i.e. I setup a Datakeeper of SciDataTool type (mag.phi_wind_stator) in a VarSimu and wasn't able to save the result file to h5 format. Saving as json was possible.

I'm not into the whole h5 thing, but (given the issue above) I think it is not equivalent to the json serialization.
So would it be possible to do the serialization first and then put the dict into a json or a h5? 
What do you think.

Best regards,
Sebastian

BonneelP, 2021-02-17:
Hello, 

We recently had some issue with h5 too. What is your version of h5py? We plan to enforce it to 2.10.0 (we had some issues with version 3.X) 
Can you also send your error message ?

If I remember well, the serialization is different, in particular we need to "flat" the list. So we can't use the same method :/

Best regards,
Pierre

SebGue, 2021-02-17:
Hello Pierre,

my version still is 2.1., since I haven't checked the requirements for a long time :-/

The error occurs in hdf5.py. The problem is, I think, that the datakeeper list of DataND is converted to a numpy array to store it. The expected type for the array data is maybe float or something. But in my case it is (non-serialized) DataND, i.e. 'variable' is of type DataTime.

[image]

I thought I read somewhere, that hdf5 could be serialized with json. Maybe I will find that source again.

Best regards, Sebastian

BonneelP, 2021-02-17:
Hello Sebastian,

We had the same issue few days ago in fact ^^' The issue is that DataKeeper.value is defined as a 'list' but in fact when creating DataKeeper of SciDataTool object, we need to handle it as "list of pyleecan object". The issue is that the content of the list is not converted to dictionary.
We solve it by enforcing the as_dict method of DataKeeper with the following code:
[image]

I will add this correction to my PR #322.

Best regards,
Pierre

BonneelP, 2021-03-09:
Hello Sebastian,

The correction was included in the PR #318, can we close this issue or do we need some further work ?

Best regards,
Pierre

SebGue, 2021-03-09:
Hello Pierre,

I think you can close this issue. But allow me some more question before :-)
Is there any other reason to use hdf5 than file size?
Best regards, Sebastian

BonneelP, 2021-03-09:
Hello,

json is more meant to save files that you can read/edit. Matrix are not really readable in json and each value is saved as a string+" ," which is not really efficient. h5 save everything in binary this is why we save so much space. For now we save simulation and machine in "flat json" and output in h5. But we can investigate other saving format if needed. 

Best regards,
Pierre

SebGue, 2021-03-09:
Hello Pierre,

I really like the readability of json, but as you noted the required space is ...
So I did some quick test last week and just zipped the json. The size of the zip was even lower than the h5. (I know hdf5 also has the ability to additional compression, so this is not completely fair.)
The point is that it would be pretty easy to implement gzip (or any other availible compressor), save a lot disk space and keep readability. Also the way of serialization uptill json.dump would be the same, so issues as the above would not happen.
So if you like that idea I would open an issue. Otherwise I would just add some small modifications to save/load to be able to 'plug in' my own code.

Best regards, Sebastian

BonneelP, 2021-03-09:
Hello, 

We never tried that so I think it is worth a dedicated issue :) With multi-simulation of multi-simulation we can expect to have very large output objects. So we can investigate a "is_compression=False" argument to the save method maybe.

Best regards,
Pierre

closed 2021-03-09 by SebGue