Eomys/pyleecan #334: [NF] Full magnet rotor

BonneelP, 2021-03-12:
Hello all,

We recently received a request to design a machine whose rotor is a cylindrical magnet (without lamination). This is an interesting design challenge for pyleecan. Here is a proposal on how to do it (feedback welcomed as usual):

First I would like the solution to use the existing MachineSIPMSM and LamSlotMag object for convenience (to avoid to create a dedicated machine type/GUI and to be able to reuse the existing models). 
With this constraints, my idea would be to introduce a new Slot (SlotM17) that would correspond to an half disk magnet (north and south). This slot won't have any property and directly use the radius of the lamination. I would enforce the following:
- Internal rotor
- rotor.Rint = 0
- rotor.p = 1
- rotor.slot.Zs = 2
- No notches, no vent, no bore shape
- rotor.L1 = rotor.magnet.Lmag
- no shaft

Regarding the slot methods:
- build_geometry: returns an empty list (cf lamination build_geometry)
- check: constraints above
- comp_angle_opening: pi
- comp_height = comp_height_active = Rext
- comp_surface = comp_surface_active = pi * Rext**2 /2
- get_surface_active: returns the half disk

Regarding the LamSlotMag methods:
- build_geometry: 
     - In LamSlot, get_bore_desc returns 2 "Slot" of pi radius without other bore lines. Then we add a check, if there is no bore lines, there is no surface to return.
     - In LamSlotMag : we call get_surface_active to get the magnet disk
- comp_angle_d_axis: Ok
- comp_surfaces: Lamination surface is defined as cylinder - slot so S=0 and the Magnet surface is Zs=2 * pi * Rext**2 /2  (cf slot comp_surface / comp_surface_active)
- comp_volume, comp_mass: Uses the surfaces => Ok
- comp_radius_mec => (Rint, Rext) Ok
- comp_periodicity => 1/2 Ok
- plot : remove "Rotor" from legend if no surface have the corresponding label. Otherwise uses build_geometry so Ok
- comp_height_yoke => 0 (we need to check if it doesn't create any problem since it would be the first "machine without yoke")
- comp_radius_mid_yoke => 0 (same)
- get_Zs / get_pole_pair => Ok

Regarding the magnet: as we use the LamSlotMag object nothing should change (we have a dedicated material and all to define magnetization). 

Then FEMM should be Ok since we will draw the magnet surfaces only (since there is no surfaces for the lamination).

I will try to implement this solution and see how it goes.
Best regards,
Pierre

SebGue, 2021-03-12:
Hello Pierre, is there some sketch of the rotor. BR

BonneelP, 2021-03-12:
Hello,

If I understood well it would look like that:
[image]
Only magnet, no lamination

Best regards,
Pierre

SebGue, 2021-03-12:
Hello Pierre,

I wouldn't spend too much affort on this and just cheat :-) I.e. if the lamination is small enough there shoul be no difference in the results.
What I recognized from this simple test machine is, that we should add some checks and corrections to W0 and Wmag (of polar magnet and maybe others too), Ideally Wmag <= W0 <= pole pitch. But due to round off errors it may be hard to exactly hit 100% pole pitch. So users should be able to set W0 and Wmag slightly larger than pole pitch to set 100%. Or we introduce some discretization, e.g. 0.1° (of course in rad).
Acctually there eighter will be a small gap (with high element count) or an boundary error (if W0 > pole pitch).

[image]

I can also post the file of this machine, if you need it. 

Best regards, Sebastian

BonneelP, 2021-03-15:
I won't push further this issue so I close it for now. I may come back to it later if someone find some issue with SlotM17.

closed 2021-03-15 by BonneelP