Eomys/pyleecan #344: [CO] Label Documentation

BonneelP, 2021-03-16:
Hello all,

In this issue we will gather all the discussions and important things to have in mind to define the label rules in pyleecan. I propose to forget the current implementation and only discuss about what should be the standard. Once the standard defined once and for all, we will update the code. 

First here are some discussions regarding labels that I found (some may be missing): #212, #195, #253

Now we need to define what is a label in pyleecan, when and why we use them. For now I see three label usages that would need to be included in the documentation:
- lamination label: We want to be able to define machines with N laminations. Then we want to introduce "lamination labels" as key for current dict for instance. cf issue #212
- surface label: this is the most important one to normalized, these labels are created in build_geometry methods and used (for instance) in FEMM to assign the proper property to the surfaces.
- line label: surfaces are defines with lines, we use label on the lines to indicate that this line can be used for boundary conditions or that it a contact line (StructElmer), we also thought at some point to indicate the magnet bottom lines but it was not needed. 

Several important aspect about label manipulation:
- We want to "decode" the label: we need to be able to conveniently use "split" and "in"
- We may want to avoid having too long labels (Elmer 30 character limitation)
- We can add other property to the surfaces if needed (as for simulation index and layer)
- To be able to study default it can be interesting to have unique surface label (different magnetization on magnet R2_T0_S8
- It can be interesting to order the different part of the label in a consistent order (more general to less general, lamination first, index last)

Regarding lamination label:
- I would rename it to "Stator0" instead of "Stator_0" (or maybe Stator-0), to be able to use "_" as the main separator (same logic for R0_T0_S0)
- For two lamination machine we still use "Stator0" and "Rotor0" to be consistent (No "Stator")
- There is a get_label(is_add_id=True) method in Lamination. is_add_id=False returns "Stator" or "Rotor", is_add_id=True requires that the lamination is within a machine (otherwise exception), and returns "Stator0". The "index" definition is the one in #212 (inner to outer, split rotor/stator) according to Machine.get_lam_list
- Machine also have the methods: get_lam_dict (key:label, value:obj), get_lam_list_label (["Stator0","Rotor0",..."Stator1"]

Regarding surface label:
- The organisation would be as follow: <Lamination label>_<surface type>_<index> or <Lamination label>_<surface type> and index as a second string property.
- surface type can be: "Winding", "Magnet", "HoleMag", "HoleVoid", "Ventilation", "Lamination"
- <Lamination label> is always return by a get_label(is_add_id=True) 
- To have consistent "split length", the Lamination surface label would be "Stator1_Lamination_0" (maybe one day lamination with several parts ?)
- index is "RX-TY-SZ" so that len(label.split("_")) == 3
- index for winding with this convention https://pyleecan.org/winding.convention.html
- index for HoleMag / HoleVoid RX-TY-SZ correspond to hole[X].magnet_Y in the "slot" Z (TODO: schematics and Hole convention)
- index for Magnet : RX-TY-SZ  X=0, Y=index in magnet list (always 0 with new LamSlotMag def), Z="Zs"
- index for Ventilation: matches Hole definition
- I would remove the "_N_", "_S_", "Parallel" "Radial"... for magnet label: we should read the corresponding data in the object thanks to the index.
- HoleMag and not HoleMagnet to avoid error with "Magnet" in label. 
- Ventilation and HoleVoid to be able to define different material
- No distinction between Winding and Bar in label

Regarding line label:
To be continued....

Feedback very welcome 
Best regards,
Pierre

BonneelP, 2021-03-19:
Update: adding some ideas on Line label. @SebGue do you have the full list of possible/useful label for lines ?

SebGue, 2021-03-19:
Sorry I had no time to anwser for now, but I already read your issue :-)
I think we should detail the 'why we need labels' some more. (Human readability would be one more point?!)
On the technical side, there should be one central instance that decode/encode the labels and also anwsers such questions as 'if label in xxx'.

BonneelP, 2021-03-23:
Update adding why and how

BonneelP, 2021-03-25:
Hello,

Tomorrow I should investigate the pending PR, run/correct some tests and then I may start writing the corresponding doc in pyleecan-doc to rephrase/reorganize all these proposals.

Best regards,
Pierre

SebGue, 2021-03-26:
Hello Pierre, sorry I don't have too much to contribute to this issue for now. I also have no good proposal on the naming of line labels.
Do you start with the docs or the actual implementation? BR

BonneelP, 2021-03-26:
Hello, 

No problem, I will start will the documentation so you still have time ;)
Anyway the implementation part may also be long ^^'

Best regards,
Pierre

SebGue, 2021-03-26:
I was just wondering if it would be better to start with the implementation.

SebGue, 2021-03-29:
Hello Pierre,

I finally took some time and thought about the label issue. 
(Big sorry for starting from scratch here again since you may already spend some time for the docs here.)

The main question is, why do we need labels. Actually we use them to pass some informations (mainly related to geometry) to functions/methods, e.g. for assign_FEMM_surf. But do we really need them? I think the answer is no (apart from lamination labels as dict keys).
Wouldn't it be possible to introduce a parent-like property to all geometrical classes that will reference the related machine's object that the geometry was created from, e.g. Lamination, Hole, Slot, Magnet, ...
Then we could check for instances rather than 'if label in XX'. Of course we then need to transport some additional information (e.g. the RxTxZx or the mag. dir.). But all in all this could simplify things.
So what do you think?

Best regards,
Sebastian

BonneelP, 2021-04-01:
Hello Sebastian,

No problem for the long answer, I currently don't have much time as well on my side :( I have mostly prepared the doc on lamination labels for now so we can change all my proposal that's the point of these discussions :) I finally had time to take a look at your latest comments and here are some thoughts.

I agree with you on the fact that what we really need is a convenient link between surfaces and the corresponding objects. As we will need the RxTySz, I would still go for labels (with the above logic) instead of a "parent" property to gather all in one property. Then what about adding a machine method "get_obj_from_label" or a surface method "get_machine_obj" (or something like that)?
But then what should we return/ set in "parent" for:
- HoleVoid: The hole object (that can also contains magnets) or the void material ? 
- Winding: we can return the Winding object but then we need to use the RxTySz label to know which current to set (and we need the current as well). 
- HoleMagnet & Magnet: we can return the corresponding Magnet object but we can also need the RxTySz if the magnet is split in several parts.
- "Ventilation", "Lamination" should be straight forward. 

I think that the logic behind such link/method would be to remove somehow the "switch case" in 
pyleecan/Functions/FEMM/assign_FEMM_surface.py. But then we may need a "assign_surf_FEMM" in all of the corresponding objects to take benefit from this system. The idea would be to avoid going from checking labels to checking instances. But whatever the case, it seems that we will have to use the RxTySz  system and so we will need to make some label manipulations to "decode" this part (get_label(is_split=True) method)

I would also keep the labels because they are interesting when you are developing/debugging a new coupling/topology to understand what we are handling (a "parent" could also work but it may be less clear). 

Regarding the "parent" solution, it would require some class generator manipulation (maybe just on Surface.as_dict) to avoid duplicating the objects for instance but it shouldn't be too complicated. 

Best regards,
Pierre

BonneelP, 2021-04-09:
I pushed the start of the modifications on https://github.com/Eomys/pyleecan-doc/tree/label and https://github.com/Eomys-public/pyleecan/tree/label. Next I will add the Lamination method get_label and reorganize all the build_geometry to import the labels from Functions/labels.

BonneelP, 2021-04-16:
Hello all,

I still haven't much time to work on this feature but I carried on the development. I have started to changed the labels and as expected it broke the FEMM coupling and here are some realizations/modifications I'm currently testing:
- For Lamination without symmetry, the surface is a SurfRing composed of 2 Surfarce objects. I have introduced "Bore" and "Yoke" for the corresponding surface label (used when assigning the mesh properties to the lines)
- For the Sliding band, the label used to be "Airgap" now it will be (no sym) Lam-X_SlidingBand with Lam-X the closest lamination to the corresponding line. It will one day enable to have several sliding band (for machines with several rotor)
- Some surfaces are not linked to a lamination (inside of the sliding band, frame, shaft) to keep the split("_") length consistency then the lamination label become "None". The inside of the sliding band is now "None_NoMesh"
- Regarding materials, I will take the time to update the current script to make it compatible with N laminations (by renaming "Stator Iron"=>"Stator-0 Iron") and to solve the #331. But then it would raise the question should type_BH_stator and type_BH_rotor be a list ? For now I will enforce the condition on all stator and all rotor and we will see if one day someone need the modification. I will introduce the "get_machine_obj_from_label" method to help on this file.

So this issue is progressively morphing into "Improving FEMM coupling for more complex machines".

Best regards,
Pierre

BonneelP, 2021-05-31:
Hello all,

Still no time to work on this PR but I have a new proposal for "line label":
- We use self.parent to get the label of the surface to know what the line is part of
- We had "prop_dict" to Line and remove the label property. Then we can have normalized keys and values (imported from labels file) for check. For instance prop_dict["BOUNDARY"] = "Yoke_Side". We know that we want to have several property set on the same line for different purpose, it should be clearer with a dict than a list. 

Best regards,
Pierre

BonneelP, 2021-09-14:
Done in #412

closed 2021-09-14 by BonneelP