Eomys/pyleecan #356: [NF] BH Curve Extrapolation

Superomeg4, 2021-03-26:
Hello everyone,

@jgdedamas and I will start working on the BH curve extrapolation method. In particular, we would like to implement the Langevin extrapolation method. 

For the coupling with FEMM, the idea would be to gave a very large number of point based on this extrapolation as the BH curve input, such that FEMM never has to extrapolate the data itself. 

Feedbacks welcome,

BR

SebGue, 2021-03-26:
Hello Raphaël,

I think if the BH curve is well defined the linear FEMM extrapolation should be sufficient. I.e. if the curve is given up to the point where the material is completely polarized, the differential permeability will be 1. So linear extrapolation will be enough.
Rather we have to check that the polarization has settled at the end of the given curve. 
BTW: How does this Langevin method work? Do you have some refs?

BR

Superomeg4, 2021-03-31:
Hello @SebGue,

We have reason to think that the extrapolation method could play a significant role on magnetic force results. Thus, we would like to check that by ourselves. 

Different models (including Langevin method) are described in this paper:
https://www.infona.pl/resource/bwmeta1.element.baztech-b5337e23-d5ad-4315-9872-06811fb079a1

BR

SebGue, 2021-03-31:
Hello, 
so your actual goal is to find some good curve fitting function for the BH curve.
Then I will come back to you for some related issue ;-)
Best regards, Sebastian

Superomeg4, 2021-04-06:
The last PR enables to use analytical defined BH curves. In particular, the Langevin model is implemented. It uses two parameters "Bs" and "a". In the following example, the parameter "a" is modified:

[image]
[image]

I have also started a "fitting" algorithm: from the input BH_curve data (from the previous workflow), the user can activate a fitting process. Then, the parameters of the model (such as previous Langevin's model) are fitted in the least square sense. 
This fitting process enables advanced extrapolation methods (in addition to the linear extrapolation from FEMM).