Eomys/pyleecan #368: Tests reorganization

helene-t, 2021-04-07:
Hi everyone,

We would like to better organize the tests in pyleecan, in terms of **folders**, **script names** and **markers**. The idea is to be able to better assess which parts of the code are covered, and avoid redundant testing. The main goal is to better use pytest markers to automatically **scan** the test scripts, using for example pytest-excel package (`pytest --excelreport=report.xlsx` creates a very handy xls file, including markers). A second step would be to merge/remove/add new tests according to the results of this scanning.

Hereafter is the new organization proposal:

- in the Validation folder, remove the Simulation level, keep the organization in folders by physic
- keep other folders as such (Classes, Functions, Methods, etc)
- rename test scripts: remove machine, "EM" notation (redundant with Magnetics folder for example), to focus on **main tested feature(s)** (e.g. periodicity, parallelization, etc)
- regroup identical tests with different machines in same script
- rename test functions (within each test script) to avoid repetitions and focus on what differenciates them within the script
- improve pytest markers use:
    - remove markers which are redundant with folder organization (e.g. METHODS)
    - **add machine markers**
    - **add model markers** (e.g. MagFEMM)
    - **add feature markers** (e.g. "periodicity", "parallelization", "Single_OP", "VarLoadCurrent", etc)
    - keep software related markers, but not mandatory since they become redundant when a corresponding model is added
    - new categories of duration markers:
         - **long_10m** for tests which take more than 10 minutes
         - **long_1m** for tests which take more than 1 minute
         - **long_5s** for tests which take more than 5 seconds

In a second step, we could add a "star" marker for a short list of tests which optimizes the feature coverage vs execution time, to be executed before any pull request.

We are aware that this new organization will result in a multiplication of markers, but we think it will improve the testing flexibility and knowledge of our coverage.

Please feel free to suggest other organizations. We will start working on updating the existing tests soon, including an update on pyleecan website to make sure that new tests follow the same standard.

Best regards,
Helene

BenjaminGabet, 2021-04-16:
Hello everyone,

I did a script to make the excel report file of the tests handier and more readable for the markers. The script is in the Test folder and his name is _excel_report_post_process.py_. The script should be called after the pytest-excel report and the report must be in the Test folder with the following name : _report.xlsx_.

Once the file is correctly placed, the script can be called. First, the report file will be converted in .csv and after that, a new file will be created with the following name : _converted_report.csv_.

Here is some screenshots:

[image]

to:

[image]

Best regards,
Benjamin

BonneelP, 2021-04-20:
Closed with #369