Eomys/pyleecan #379: Electrical Equivalent Circuit for PMSM with Stator Damper Winding

Sijie-NI, 2021-04-29:
This project concerns a new electrical equivalent circuit (EEC) of a silent permanent magnet synchronous machine with a damper winding in stator slots. This project is based on a PhD thesis of the laboratory LSEE in the University of Artois "Damper winding for noise and vibration reduction of Permanent Magnet Synchronous Machine"

The EEC is separated into 2 parts, on the d-axis and on the q-axis:

d-axis:
[image]

q_axis:
[image]

SebGue, 2021-04-29:
Hello Sijie-NI,

very interesting project. Could you give an explaination on the EEC or a link to the related thesis?

Best regards, Sebastian

Sijie-NI, 2021-04-29:
> Hello Sijie-NI,
> 
> very interesting project. Could you give an explanation on the EEC or a link to the related thesis?
> Why is there a capacitance in series?
> 
> Best regards, Sebastian

Hello Sebastian, 

Thanks for your interest. 

I am PhD student at the University of Artois, who is working on this PhD thesis. The explanation will be completed later. I am learning how to model this machine in Pyleecan. We add this capacitor to form an RLC circuit in the damper winding, whose resonance frequency decides which harmonic could be reduced by damper winding. I have started my thesis in the laboratory LSEE. But I recommend papers and the thesis of my colleague Grégory Bauw, who applied this passive method in an Induction Machine, "Design of an induction machine with damper windings for noise and vibrations reduction". 

LSEE website:
http://lsee.univ-artois.fr/

Thank you very much

Sijie NI