Eomys/pyleecan #381: Issue about saving FEMM project

nowis1238, 2021-05-03:
Hello everyone,

I have already contact Pyleecan developer about this issue by mail but I think it is better to post my isssue directly on github. This way, all the community can benefit from the solution.

For several months, I have tried to add pyleecan library to my project. Now, I am very close to my objective but I have still this annoying problem I cannot solve.

Here is the problem : 

When I try to run a Femm simulation using pyleecan I have this error : 

Traceback (most recent call last):

    out_femm = simu_test.simu.run()
  File "...\pyleecan\pyleecan\Methods\Simulation\Simulation\run.py", line 52, in run
    self.run_single()
  File "..\pyleecan\pyleecan\Methods\Simulation\Simu1\run_single.py", line 31, in run_single
    self.mag.run()
  File "...\pyleecan\pyleecan\Methods\Simulation\Magnetics\run.py", line 24, in run
    out_dict = self.comp_flux_airgap(output, axes_dict)
  File "...\pyleecan\pyleecan\Methods\Simulation\MagFEMM\comp_flux_airgap.py", line 96, in comp_flux_airgap
    FEMM_dict = draw_FEMM(
  File "...\pyleecan\pyleecan\Functions\FEMM\draw_FEMM.py", line 220, in draw_FEMM
    femm.mi_saveas(path_save)  # Save
  File "...\pyleecan\pyleecan\Classes\_FEMMHandler.py", line 2127, in mi_saveas
    self.callfemm("mi_saveas(" + self.quote(self.fixpath(fn)) + ")")
  File "...\pyleecan\pyleecan\Classes\_FEMMHandler.py", line 66, in callfemm
    raise Exception(ErrorMsg)
Exception: error: Couldn't write to specified file.
Perhaps the file is write-protected?

Here is what I' have tried to understand the problem.
 
As suggested, I try to find if I have a problem to write file in the directory, Apparently not : 

- I tried to create and write a file in the directory in Python and I have no problem.
- I tried to save manually the Femm project and to use Lua console with exactly the same path and it works perfectly.
- I tried to force the Femm project to be saved in another directory and I have the same problem.
- I tried many other typo with "/", "//", or even "\" instead of "\\" but it does not solve the problem.

Here is what I get when I use output.get_path_result :
...\pyleecan\pyleecan\Results\mag_simu_test

I hope someone can help me because I am quite stuck for the moment.

many thanks,

Simon Morel

BonneelP, 2021-05-03:
Hello Simon,

I never had such kind of issue so it is really strange :/ 

To help you solve that, let's start with some usual context questions:
- Are you using pyleecan from pip or from a fork/git clone ? 
- Which version of pyleecan/python/femm are you using ?
- Your main path "..." is it in the python site-package directory ? 
- Which machine were you trying to draw (I don't expect it to be the issue but does it work with the simulation in the Tests folder for instance) ? 
- Is there any message in the console / log file (%APPDATA%/pyleecan/Pyleecan.log) ?

If you are using a fork/git clone version of pyleecan, can you add the following line just before the call to mi_saveas dans Functions/FEMM/draw_FEMM.py ?
```python
output.get_logger().debug("Saving FEMM file at: " + path_save)
femm.mi_saveas(path_save)  # Save
```
To make sure that the path is correct. 

What is the name of your simulation ? If it begin with a "t" the issue can be that the path contains a "\t". In Methods/Simulation/MagFEMM/get_path_save_fem.py we create the path_save with a "join" but without a call to ".replace("\\", "/"). 

Best regards,
Pierre

nowis1238, 2021-05-04:
Hello Pierre,

thank you for your help. Here are the answers of  your questions :

- Are you using pyleecan from pip or from a fork/git clone ?

   I tried both and each solution returns the error.

- Which version of pyleecan/python/femm are you using ?
    pyleecan 1.2.1
    python 3.7.10
    femm 4.2

- Your main path "..." is it in the python site-package directory ?

   no it is not.

- Which machine were you trying to draw (I don't expect it to be the issue but does it work with the simulation in the Tests folder for instance) ?
    When I run the dbugger in my project I can save and run manually the project in FEMM gui and it works perfectly.

- Is there any message in the console / log file (%APPDATA%/pyleecan/Pyleecan.log) ?

2021-04-08 18:40:31-  DEBUG: Updating USER_DIR from version No_version to version 1.2.1 [Pyleecan.GUI]
2021-04-08 18:40:31-  DEBUG: Updating USER_DIR Machines in C:/Users/simon.morel/AppData/Roaming/pyleecan\Machine [Pyleecan.GUI]
2021-04-08 18:40:31-  DEBUG: Updating USER_DIR Materials in C:/Users/simon.morel/AppData/Roaming/pyleecan\Material [Pyleecan.GUI]
2021-04-08 18:40:31-  DEBUG: Updating USER_DIR Plot in C:/Users/simon.morel/AppData/Roaming/pyleecan\Plot [Pyleecan.GUI]
2021-04-08 18:40:31-  DEBUG: Updating USER_DIR GUI in C:/Users/simon.morel/AppData/Roaming/pyleecan\GUI [Pyleecan.GUI]

- If you are using a fork/git clone version of pyleecan, can you add the following line just before the call to mi_saveas dans Functions/FEMM/draw_FEMM.py ?

...\pyleecan\pyleecan\Results\mag_simu_test\Femm\default_machine_model.fem (where "..." is the same path as previous "...").

BonneelP, 2021-05-04:
Hello Simon,

In the "..." part is there any special symbol or quote maybe ? If you managed to save the file with a breakpoint then the FEMM coupling is not the issue but maybe the command itself. It the traceback you send there is:

> File "...\pyleecan\pyleecan\Classes_FEMMHandler.py", line 2127, in mi_saveas
> self.callfemm("mi_saveas(" + self.quote(self.fixpath(fn)) + ")")

can you check with a breakpoint what is happening here ? Maybe the generated command is wrong.

In pyleecan\pyleecan\Results\mag_simu_test you should also have a log file dedicated to your simulation. Any useful information in it ? By the way, it could be interesting to check if the folder "pyleecan\pyleecan\Results\mag_simu_test\Femm\" exist (it should be empty since the error occurs when saving the fem file). If the FEMM folder doesn't exist, it can explain why the save command fails. 

Best regards,
Pierre

nowis1238, 2021-05-06:
Hello Pierre,

According to me there is no special symbol int the path. But I tried to force the FEMM project to be saved in a path as simple as Documents and I achieved to save the project. I work on a network drive and it appears that if the Lua command work in the Lua console in FEMM, the pyFEMM one doesn't.

The path shape of looks like : 

"\\drivename.local\folder\...\folder\pyleecan"

I see nothing wrong in the log file except the process is not finished : 

2021-05-04 14:35:10-  DEBUG: ################################### [mag_simu_test]
2021-05-04 14:35:10-   INFO: Starting running simulation mag_simu_test (machine=default_machine) [mag_simu_test]
2021-05-04 14:35:10-   INFO: Starting Magnetic module [mag_simu_test.Magnetics]
2021-05-04 14:35:10-  DEBUG: Drawing machine in FEMM... [mag_simu_test.Magnetics]

best regards,

Simon

nowis1238, 2021-05-06:
I actually reuse pyleecan using pip because the library is stored on a local drive. Last time I used pyleecan with pip I forgot I had an environment variable for pyleecan, thus I was still using the cloned project.

Now It looks to work fine but I am still interested to know if we can use pyleecan or pyfemm when the library is in a network drive.

thanks,
regards,

Simon

Eomys, 2021-05-06:
Hello Simon,

Great to know that you managed to make it work :)
Sadly, I have no idea why it can't work on network drive :/

Best regards,
Pierre

BonneelP, 2021-05-06:
Hello Simon,

Great to know that you managed to make it work :)
Sadly, I have no idea why it can't work on network drive :/

Best regards,
Pierre