Eomys/pyleecan #383: Permission error while running tests on Windows 10 and Anaconda

emap78, 2021-05-18:
Hello,

I installed pyleecan as suggested, but when I run

`python -m pytest -m "not long"`

I get the following error:

[image]

If I launch the console with administrator privileges, I get the error on a different folder:

[image]

Maybe everything is ok with the installation, but how can I run the test correctly).

Thanks

BonneelP, 2021-05-18:
Hello,

Thank you for your interest in pyleecan. 
How did you install pyleecan ? From pip install or by getting the code from a fork / git clone ?
The tests are meant to be run with the fork version (maybe we need to update the documentation).

To run the test, the command is now:
`python  -m pytest <./path/to/pyleecan/folder> -m "not long_5s"`
(The path need to be the folder containing the Tests folder) We have change the marker from "long" to "long_5s". Can you send me the link where you found this command so I can update it ? 

Best regards,
Pierre

emap78, 2021-05-18:
Hi,

I installed pyleecan from pip. 

I tried with 
`python -m pytest <./path/to/pyleecan/folder> -m "not long_5s"`
where in my case <./path/to/pyleecan/folder> should be C:\Users\<USERNAME>\Anaconda3\envs\pylecaan_env\Lib\site-packages\pyleecan

but now it says "no tests ran in ...s"

EmileDvs, 2021-05-18:
Hello emap78,

As @BonneelP said, you cannot run test on the pip installed version. You'd have first to clone pyleecan from github in an empty folder, let's say in a new folder C:/pyleecan, and then run the command you mentionned at this folder path.

Best regards,
Emile

emap78, 2021-05-19:
Hello thanks for your support, so I just think the installation guideline needs an update as @BonneelP said.

Note also that for launching the GUI, the command `run -m pyleecan` does not work and should be replaced with `python -m pyleecan` (on Windows).

Thanks again

gforti84, 2021-11-10:
What is the expected result?
[image]
?

SebGue, 2021-11-10:
Hello @gforti84 

the tests are mainly to verify that pyleecan works properly, e.g. if you did some code changes. 
As meantioned before these test are only avialable if you forked pyleecan from github.

Best regards, Sebastian

BonneelP, 2021-12-03:
Is there anything else to discuss in this issue or can we close it ?