Eomys/pyleecan #394: Magnet Slot 57 fails for certain geometry

carbon-drive, 2021-05-27:
Dear all,

I tried using Slot 57 for a machine and it doesn't work with femm. The problem is the position of the "material properties point" of the "Rotor Iron". In my case it ends up inside the shaft.
[image]

I would be willing to fix this myself and already found the class HoleM57.py but honestly I was not able to find the definition of the geometry there. I am aware I should watch the webinar 3 and do the corresponding tutorials first before attempting to change slot geometries. So once I did this and nobody was faster I will try to implement a better position. My suggestion would be 1µm outside of one of the magnets so for example from "HoleMagnet_Rotor_Parallel_N_R0_T1_S0" just H2*0.5+1µm in direction of the magnetization:
[image]

BonneelP, 2021-05-27:
Hello,

Thank you for noticing that :) 
The geometry modeller of pyleecan create a list of surfaces corresponding to the different objects of the machine. Then the "rotor iron" is not Hole related but lamination related. So the definition of this point is made here:
https://github.com/Eomys/pyleecan/blob/master/pyleecan/Methods/Machine/LamHole/build_geometry.py line 54
The issue seems to be that the "yoke mid radius" is wrong and the definition is here: 
https://github.com/Eomys/pyleecan/blob/master/pyleecan/Methods/Machine/LamHole/comp_radius_mid_yoke.py

Can you check that you set rotor.is_internal = True ? Otherwise, the error can be in Hole.comp_radius: the point of this method is to get the minimum and maximum radius of the points of the Hole. To define the position of the reference point we need to take into account that we can have several holes (and also vents that are not taken into account but they should also have the comp_radius method). The point of comp_radius_mid_yoke is to find a radius where we know there are no holes.
Looking forward for your PR, don't hesitate if you need any further assistance ;)

(by the way it make me realized that I forgot to correct the reference point in my latest PR, thanks :) )

Best regards,
Pierre

BonneelP, 2021-09-17:
I have fixed a typo in #435 that should correct it (to check)

BonneelP, 2021-09-24:
Hello,
In fact the error was in Hole57.comp_radius. It will be corrected in #431 
[image]

Thak you for spoting this error :)
Best regards,
Pierre

closed 2021-09-24 by BonneelP