Eomys/pyleecan #400: Can't install pyleecan using pip install

azdinfaouzi, 2021-06-09:
I tried installing pyleecan using the command :
pip install pyleecan 
it gives me an error of something concerning iniconfig, i solved that probleme by installing the module through github install:
first you clone the repository:
git clone https://github.com/Eomys/pyleecan.git
then go to the location and run setup.py : 
sudo python setup.py install

thalesmaoa, 2021-11-10:
```pip install git+https://github.com/Eomys/pyleecan.git```

BonneelP, 2021-12-03:
Hello,

I managed to install pyleecan 1.3.7 (released last week) with pip and run the GUI. Do you still have some issue with pip or can we close this issue ? 

Best regards,
Pierre

thalesmaoa, 2021-12-03:
Using an workaround for Linux. You can close.

BonneelP, 2021-12-03:
Thanks you for your return

closed 2021-12-03 by BonneelP