Eomys/pyleecan #406: BUG: load abstract class / init by dict

SebGue, 2021-06-29:
Hello,

there is a (maybe rare) load issue. 
Since the load function uses the init by dict functionallity there will be an assertion error for abstract classes.
E.g. from EEC class:
```
def __init__(self, init_dict=None, init_str=None):
        ...
        if init_dict is not None:  # Initialisation by dict
            assert init_dict == {"__class__": "EEC"}
```

The error will occur since the init_dict will not only have the '__class__' key/value pair, but also the '__version__' and '__save_data__' keys.

So there should rather be the check for `assert init_dict['__class__'] == 'EEC' ` or something similar.

Best regards, S

BonneelP, 2021-07-01:
Hello Sebastian,

Sorry for the long time without answer. I don't know how you managed to spot this but well done :) 
I have corrected it in the following commit 1faedde4b24acc6361fa1fdd4e980eaec4ca3a62 (directly on master)

Best regards,
Pierre

SebGue, 2021-07-02:
Hello Pierre,

you are welcome. I'm working on some kind of (internal) project managment tool. Therefore I also have a dialog/tool to get an arbitary pyleecan class. So I was playing around saving and loading random pyleecan objects to the project and found this one.

BTW: I will release the class dialog / tool and some generic pyleecan object editor later.

Best regards, Sebastian

BonneelP, 2021-08-09:
Hello,

I think that this issue is solved, feel free to reopen it otherwise ;)

Best regards,
Pierre

closed 2021-08-09 by BonneelP