Eomys/pyleecan #413: [NF] DQ extension to DQH

BonneelP, 2021-08-02:
Hello all, 

We plan to extend the existing DQ transform methods to the DQH frame. 

Best regards,
Pierre

Sijie-NI, 2021-08-04:
Hello,
We can add one dimension in the transformation matrix as follows. P =2/3 for keeping the amplitude, P=sqrt(2/3) for keeping the power. Inversely P'=1 for the amplitude, P'=sqrt(2/3) for the same power.

[image]

Sijie-NI, 2021-08-05:
Hello, I share the voltage equations on the dq0 frame:

[image]

Best regards
Sijie

EmileDvs, 2021-12-09:
issue solved with #464

closed 2021-12-09 by EmileDvs