Eomys/pyleecan #419: Microsoft specific fonts used as default, and default_config_dict not changing font.

mjfwest, 2021-08-30:
It seems that Arial has been chosen as a default font, instead of a generic sans-serrif font.  Arial is not free and is therfore not installed on every platform. This generates a warning `[12:33:45] WARNING: Arialfont not available. Try: matplotlib.font_manager._rebuild()`.   (rebuilding won't work if you don't have the font)
Changing the font in `default_config_dict.py` does not fix the issue. 
`default_config_dict["PLOT"]["FONT_NAME"] = "FreeSans"`
further inspection of the codebase shows that `DEFAULT_FONT = "Arial"` is hard coded in `init_environment.py`, and also hard coded in many other plot functions and methods. 

Proposed fixes/questions:

- Outside the GUI, Is it really necessay for pyleecan to have an opinion on what font is used? 
- If it does, can it specify a generic font family such as 'sans-serif' and allow the platform to choose the installed equivalent font.
- If configuring a specific font is necessary for some reason, can the default_config_dict be used consistently?
-

mjfwest, 2021-08-30:
Would the following minimal change be accepted?  It would at least check to see what the default sans-serif font actually is, rather than assuming Arial is a default font.
```
from ..loggers import GUI_LOG_NAME
from ..default_config_dict import default_config_dict

# DEFAULT_FONT = "Arial"

DEFAULT_FONT = font_manager.findfont(font_manager.FontProperties(family=['sans-serif']))
DEFAULT_FONT = DEFAULT_FONT.split('/')[-1].split('.')[0]
DEFAULT_COLOR_MAP = "RdBu_r"

```

BonneelP, 2021-09-05:
Hello Martin,

Sorry for the long time to answer, I was in holiday. 
The point of config_dict["PLOT"]["FONT_NAME"] is to be able to select the font to use in all the graph/plot commands so it used in scripting as well. It is used to normalize the graph font for reports or publications. 

Normally to change the font you need to update the value in the main_config_dict.json file which is saved in the %APPDATA%/pyleecan folder. Changing the value in default_config_dict affect the original value when creating the main_config_dict when running pyleecan for the first time. 

In fact, we didn't put many thought when selecting our default font (we basically used the first one that we had in mind at the moment), sorry for that. You are right, we should have picked a free and universal one. Thank you for your code proposal, I will include it in the release that I should do this week hopefully. I will also take the time to check that the value used in the plot is the one in main_config_dict.json and not DEFAULT_FONT. Meanwhile, you should be able to update the value in main_config_dict.json. 

Best regards,
Pierre

mjfwest, 2021-09-06:
Thanks for your answer.  I completly agree that it is nice to normalise the graph/plot commands for consistency in reports or publications, but isn't this function built into matplotlib? I refer to the  [styles](https://matplotlib.org/stable/tutorials/introductory/customizing.html) module

If I could suggest moving your plottting styles to the matplotlib style sheet, then it can not only be used consistently in pyleecan, but any other figures that need to be generated by extension scripts, or scripts outside pyleecan can also have a consistent font. 
At the momnent, you can only get consistency within plots generated by Pyleecan?

This covers all plotting styles not just fonts, so may be better filed as a suggested improvement in a new issue? 

To close this issue: a search for `Arial` in github finds other locations, that appear to hard code arial as a font, should they also be removed? 

https://github.com/Eomys/pyleecan/blob/43d8cc0f8c7c048af3766b64c1f68304c9dceaab/pyleecan/Functions/Plot/Pyvista/configure_plot.py#L59
 
https://github.com/Eomys/pyleecan/blob/7720353229b86124361a8ee7afeb1d83c2b58c31/pyleecan/Methods/Mesh/MeshSolution/plot_deflection.py#L164

BonneelP, 2021-09-10:
Hello,

I have reorganized the plot font with your code suggestion. Thanks again for your help on this topic. 
The tests are currently running then I will create a new release. 

Regarding pyvista, it seems that we need to chose the font between "arial", "courier", "times" so I have created a new dedicated parameter in the config_dict (normally pyleecan should update your config_dict but you can delete it in %appdata%/pyleecan/main_config_dict.json)

Regarding the style module for matplotlib, if you can open a dedicated issue that would be great :)

Best regards,
Pierre