Eomys/pyleecan #421: Problem regarding installing pyleecan

Ksp-3086, 2021-08-31:
Hi,
I am installing pyleecan for the first time. I already have installed python 3.6 and microsoft visual studio as an IDE. On the site it says to type the command " pip install pyleecan". I opened the shell for python and typed the command there, it did not work. I am not that well versed in programming so is there something I am missing or doing wrong?

SebGue, 2021-08-31:
Hello KSP,

you got to run that command in your OS shell. Maybe you need administrativ rights for it to complete.

Best regards, Sebastian

Ksp-3086, 2021-08-31:
Hello @SebGue ,
I opened up command prompt on my windows and tried to run it under the administrative username. Do I need to go into a specific directory where pip or python is installed?

SebGue, 2021-08-31:
You can run it from wherever you like.

Ksp-3086, 2021-08-31:
I got an error like this in the image when I ran it in the command prompt.
[image]
It says pip is not recognized as an external command or an operable  program or batch file.

SebGue, 2021-08-31:
Okay, then you got to go to the pip install directory, since the python path seems not to be in your environment variables.

Ksp-3086, 2021-08-31:
[image]
Yup, it still did not work

SebGue, 2021-08-31:
If you go to the dir where pip is located, it should work. But i can't tell you where that is, right now.

Ksp-3086, 2021-08-31:
[image]
it shows python not installed and neither is pip. For pyleecan it was recommended to use from 3.6 to 3.8 so I chose 3.6 . I have it installed as you can see in the picture but I do not know why is cmd showing like this. Any ideas

SebGue, 2021-08-31:
I think thats because your python path is not in the path environment variable. 
You have to find your python installation directory (where the python.exe is loacted). In my case it is 'C:\Users\SebGue\AppData\Local\Programs\Python\Python38\'. There is also a 'Scripts' folder with the pip.exe
You may add that two directories to your path variable.

But my proposal would be to uninstall python and install 3.8 instead. Also make sure you use the x64 version. While the installation you have the possibility to add the path to the environment variable. So you should check this option.
Then you will be able to run python and pip from any path in the cmd.

Hope this will finally help to solve your problem.

BR, Sebastian

Ksp-3086, 2021-09-01:
Thanks for the help. Its working with python 3.8 without even going to the directory. The reason why 3.6 was not working was as it uses different commands or maybe the support is withdrawn already.

SebGue, 2021-09-01:
You are welcome. So can we close this issue?

closed 2021-09-01 by Ksp-3086