Eomys/pyleecan #422: Problem while installing pyleecan

Ksp-3086, 2021-09-02:
Hi,
I have just started to install pyleecan and I have a bit of a setback. I have already installed the pyleecan in command window on windows 10. When asked to run the command "python -m pytest -m "not long"" to check if everything is working properly, I get the following.
[image]

SebGue, 2021-09-02:
Hello Ksp,

for pyleecan to run you don't need to run the tests (i.e. the command "python -m pytest -m "not long").
Just run the GUI. The command should be 'python -m pyleecan'.

Best regards, Sebastian

Ksp-3086, 2021-09-02:
Thanks that helped