Eomys/pyleecan #423: Guidance on running the tutorial

Ksp-3086, 2021-09-02:
Hi,
I am just getting started with the pyleecan software and needed to ask a few things about running the scripts.
[image]
As mentioned this script can be run on python but I was able to get this
[image]
I have a bit of experience running programs on visual studio but this is the first time running python. I cannot write the full script as seen on the tutorial page, so do I like write in a notepad and then copy paste it?

SebGue, 2021-09-02:
My suggestion would be to use some IDE, e.g. VS Code or you simply use pythons included 'idle'.
I think there are a lot of resources on getting started with python on any IDE.
But if you got other (pyleecan related) question don't hesitate to ask.
Actually I don't know if there is also a pyleecan 'getting started'.

BR S

SebGue, 2021-09-02:
Fotgot...
The tutorials are written as jupyter notebooks.

Ksp-3086, 2021-09-02:
For the jupyter notebooks, I have to use anaconda software, right? Right now, I am practicing tutorials. I have to model an outer rotor SRM and its efficiency map for my internship. I have managed to make a JSON file in pyleecan and now I have to couple it with the FEMM software which I have also downloaded. So, I managed to put the code in MSVC. I chose an empty python module. I am confused as to where do I debug and run the code?
[image]

Ksp-3086, 2021-09-02:
This picture is from an example code on the site. Also, I need to ask where can I find syntaxes for SRM  and other stuff because I do not know much about these syntaxes

SebGue, 2021-09-03:
Hello Ksp,

I guess you are using Visual Studio, so I don't know exactly how to run, debug, etc. 
I'm using VS Code with the Python extension. Jupyter notebooks should also be possible with VS Code, but I haven't used them very often. [VS Code with Jupyter](https://code.visualstudio.com/docs/datascience/jupyter-notebooks) Just have a look at the respective VS Code python docs to get started with running, debugging, ...

Regarding syntaxes, it's all python syntax where you create objects, run methods on them, etc. You 'only' have to read the pyleecan [classes docs](https://www.pyleecan.org/pyleecan.Classes.html) to get to know, what the class properties and methods are. I know the docs are not user friendly too much. But all in all for the SRM it's very similar to what you did in the tutorials. You could also download the source code and have a look at the csv files in the 'Generator\ClassesRef\..' folders. These files are used to define all the pyleecan classes. There should also be at least one example SRM in the machines folder. 

Hope that helps a little further.

Best regards, Sebastian

Ksp-3086, 2021-09-03:
[image]
It seems all code was written in jupyter. So, now I downloaded the tutorial and ran it in the jupyter notebook. Its giving me this error. I have pyleecan installed so what is the problem?

SebGue, 2021-09-03:
You should be aware that there may be different python environments depending on how you run your notebook (e.g. see your VS figure, where there are 3 different environments on the rigth side). So the 'cmd' python where you installed pyleecan may not be the python used to run the notebook.
Therefore you have to make sure you are either using the 'cmd' (i.e. windows installed) python with pyleecan installed or you install pyleecan in your notebook servers python environment.

Ksp-3086, 2021-09-03:
[image]
So, I installed pyleecan from anaconda now, anaconda uses python 3.8 while pyleecan was installed on 3.10. It gives an error on permissions. As far as I know I am running on administrative account so how do I check permissions here

Ksp-3086, 2021-09-03:
[image]
I get errors while running the tutorial still that some files are missing. Is it normal or do I have to do a full uninstall and a reinstall again?

SebGue, 2021-09-03:
It seems to be a little messed up...
I think you should first decide which IDE you would like to use. If you are into Visual Studio there may be resouces how to use python, python environments and how to install python packages there.
You could also try VS Code, as I do :-) 
But Anaconda may also be fine (I also have it installed but don't use it often.)

Then you have to decide about which environment to use. The 'how to' depend on your IDE.
In any case you have to make sure to install pyleecan properly, i.e. installation should not show erros. Sometime you need to run the run the 'pip instal' twice or you got to install the packages with errors manually (e.g. pip install numpy). All the required packages can be found [here](https://github.com/Eomys/pyleecan/blob/master/requirements.txt).

Ksp-3086, 2021-09-03:
So, I can run the tutorials from VS code and Anaconda, if I get that right. I tried Visual studio on how to use python but did not get much help. I have never used VS Code but as tutorials are given in anaconda, I decided to use that. Right now I am using anaconda as I was unable to use Visual studio for debugging and running the code. On the pyleecan site, it asks that if using anaconda we need to add a new environment variable. https://www.pyleecan.org/get.pyleecan.html
I did that but still I am unable to open the GUI of pyleecan

Ksp-3086, 2021-09-03:
I have uninstalled everything(anaconda and all python) and started from only anaconda. It still gives the same errors probably some leftovers like cache. Any suggestions how to do that. Forgot to mention that after I changed the environmental variables, the anaconda app wouldn't open. I tried doing the steps from the site https://docs.anaconda.com/anaconda/navigator/troubleshooting/#:~:text=If%20you%20cannot%20launch%20the,continuum.&text=Then%20relaunch%20Navigator%20from%20the,%2C%20terminal%2C%20or%20Anaconda%20Prompt. but none worked. Even after the reinstall it isn't working

SebGue, 2021-09-03:
Sorry I can really help you with anaconda since i don't have much experiance with it.
If you don't get it running, you should try VS Code.

closed 2021-09-03 by Ksp-3086