Eomys/pyleecan #425: Problems running pyleecan on Anaconda

Ksp-3086, 2021-09-03:
[image]
When I installed pyleecan from the anaconda prompt, I got this. I don't know what is this supposed to mean but then I get this when trying to run pyleecan GUI
[image]
As mentioned on the site, https://www.pyleecan.org/get.pyleecan.html, I changed my environmental variables but then too I get the same errors. Furthermore, after changing the environment variables, anaconda navigator has stopped opening. I had too uninstall and reinstall it but it still doesn't open.

SebGue, 2021-09-03:
For the above error you should downgrade PyQt to maybe 5.15.3 or even 5.15.2 and rerun pip install pyleecan, since installation of requirements was incomplete because of the error.
As I told, sometimes it is necessary to manually install some of the required packages. 

But it could also be some issue with the requirements.txt of pyleecan. I will check this later.

SebGue, 2021-09-03:
... I also think you should delete the environmental variable, since I can start the GUI (with Anaconda) without it ever set.

Ksp-3086, 2021-09-03:
How do I downgrade the version, like type in cmd "delete PyQt" and then "install PyQt 5.15.3"?

SebGue, 2021-09-03:
You can do 'pip install --upgrade PyQT==5.15.3', but this is not a pyleecan related question.

You just open the requirements.txt file in you favorit editor, to which other packages are needed for pyleecan (and which version).

Ksp-3086, 2021-09-03:
[image]
So I am unable to install any older versions as it says they don't exist. Anytime I have reinstalled pyleecan through anaconda it installs the latest version of pyqt5

Ksp-3086, 2021-09-03:
[image]
So I was able to downgrade the versions of both pyqt and pyqtwebengine. Then I uninstalled pyleecan and reinstalled it and it still shows the same error

SebGue, 2021-09-03:
But you were able to start the GUI before on cmd or VS?!
From the error message it seems that this is some PySide2 / Anaconda issue (not related to pyleecan). 
Can you check if shiboken2 is installed e.g. with 'pip list'.

Ksp-3086, 2021-09-03:
[image]
yeah before, when I installed pyleecan directly from python, I was able to run the GUI. I needed anaconda to run the jupyter notebooks. shiboken2 seems to be installed

SebGue, 2021-09-03:
One last thing you could try is to setup a new empty environment in anaconda and try installing pyleecan there.
Otherwise you should find someone who know anaconda better than me.

Ksp-3086, 2021-09-03:
Can you elaborate on new environment? Do you mean a new path? Yeah you mentioned in the last thread that you did not know anaconda so I started a new one in the hopes that someone may respond. No one else did so I can only hope someone comes looking here otherwise I will be unable to run the jupyter notebooks.

SebGue, 2021-09-03:
In anaconda there is a side menu 'Environment' where you can setup a new one. 
That also the point about environments, you can switch between them and install   only the packages you need for some purpose. So you can avoid cinflicts between packages, etc. You can even install different python versions that way.

Ksp-3086, 2021-09-03:
yup tried that I still have the same issue. Thanks for your help anyways. So, should I just close this thread and start a new one?

closed 2021-09-03 by Ksp-3086