Eomys/pyleecan #427: Working of jupyter notebook in vs code

Ksp-3086, 2021-09-04:
[image]
The code seems to run fine in vs code but it doesn't plot the machine as seen in the figure in tutorials. If I run the same code by copy pasting in python terminal, it plots the machine so why not here

mxgnsr, 2021-09-06:
Hi, 

running it with the "%matplotlib notebook" statement did not work for me either. 
Can yout try "%matplotlib inline"?

BR
Max

Ksp-3086, 2021-09-06:
Yeah thanks it did work although it gave a warning sort of "UserWarning: Matplotlib is currently using module://matplotlib_inline.backend_inline, which is a non-GUI backend, so cannot show the figure.
  fig.show()".
Can you tell me why this worked and why is this warning be shown? I can't see " fig.show()" in the code.

mxgnsr, 2021-09-07:
According to this [Issue](https://github.com/microsoft/vscode-jupyter/issues/4056), vscode does currently not support the notebook-mode for matplotlib. Inline-mode seems to work, although I personally don't get that warning.
The `fig.show()` statement is part of the `machine.plot()` function.

Ksp-3086, 2021-09-07:
ok thanks for the help

closed 2021-09-07 by Ksp-3086