Eomys/pyleecan #428: Problems in SRM modelling

Ksp-3086, 2021-09-06:
I have modelled an SRM in pyleecan GUI. However, I need to run the FEMM simulation to get iron and copper losses and torque results. I have the following problems as of now ( as I am new to python and pyleecan both):

1. How do I define the phase current for switching circuits?
2. How do I calculate losses (both iron and copper losses, if I can get overall losses that would be great too")
3. How do I output the data to a dat file so that I can use the data for future references and in other software like MATLAB?

SebGue, 2021-09-06:
Hello Ksp,

from your question I guess there is no such tutorial yet? Would you mind to create a tutorial on that if I guide you through?
Best regards, Sebastian

Ksp-3086, 2021-09-06:
Yeah, sure that would be helpful. I do have a slight idea on how to do the step functions (as SRM works on that) but I have not worked on python programming, so yeah it would be a huge help.

SebGue, 2021-09-07:
Okay, then I think it would be best to start with [the tutorial on FEMM simlations](https://www.pyleecan.org/tuto_Simulation_FEMM.html). There you already got individual phase currents defined as function of time. You only need to replace them.
Since I don't know your actual current waveform there may be several ways to define them, depending on your pereference (e.g. with fourier series or step functions or ...). In any case I would recommand to define a function that returns the current waveform, so you can reuse it and your code is more readable. E.g. 
```
def rect_wave(time, freq, width, phase):
    current = ...
    return current
```
Then with the currents you can run your simulation and check if you get the expected torque values.
We will address the other points after that.

OT: Do you have further experiance in SRM expecially in 6 phase machines for sinewave drives?

Ksp-3086, 2021-09-07:
Ok, I have gone through the tutorial and yeah I think I need a rectangular wave form as a function of time. I just have some problem defining the rectangular waveform. I do not have much experience in SRM and 6 phase machines for sinewave drives.

Also, pyleecan does not let me define more or less than 3 phases for SRM, it always gives out an error. I am also a bit confused what does pole pairs mean here? As far as I remember pole pairs meant N-S pairs, but here that does not seem to be the case

SebGue, 2021-09-07:
Oh, it's even easier than I thought. There is a dedicated numpy function for [rectangular signals](https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.square.html).
Aside from my question, do you need other than 3 phases now? Maybe you can post your current code to make your questions more clear.

Ksp-3086, 2021-09-07:
[image]
I think I do not need more than 3 phases for now but it does give an error of I need to increase it in the future. The error is
[image]
Also about the pole pairs take a look at the images where I used 2,6 and 12 pole pairs. Aside from the change in position of the windings there was not much change which is why I am confused.
[image]
[image]
[image]
The dedicated numpy function is for square waveforms as it is shown on the site as I can see and not for the rectangular one. For this tutorial, I think we would need the current in each phase to be a positive constant value for 1 cycle and 0 for the next two. So that is somewhat different

Ksp-3086, 2021-09-08:
Hi @SebGue ,
I got the function for the rectangular wave and I put it in. As I am new to python, how can I return the value needed as you can see in the code below.
[image]

SebGue, 2021-09-08:
For the multiphase error, I have to check if the winding generator is able to generate such winding. Anyway you can allways setup an user winding with phase count other than 3.

The windings for the different pole pair numbers seem to be valid. What did you expect there? 

Side note: The machine with 12 pole pairs would not be used with 'normal' machines due to its low winding factor. Is this a valid pole/slot combination in SRM?

If you offset the numpy rect function and use a duty cycle of 33% it should also work.
But you fourier series is also okay. You only got to init the x variable e.g. with numpys zeros or zeros_like functions.
Further for indexing in python you need to use square brackets [ ] 

Hope that help you for now.

Ksp-3086, 2021-09-08:
How do I setup phase count other than 3? Is it by changing the code in json file?

Visually, the pole pairs kind of look the same. I think I am just confused in that part like, the dotted represents current going inwards and the other one outwards. If I set pole pairs to 2, then there should be only 2 N-S pairs but there seem to be 3 for each phase (like 3 dotted and 3 checkered)

12 pole pairs is not valid, I was just playing with the GUI to see different outcomes or if it gives an error.

Ksp-3086, 2021-09-08:
[image]
So, what does this error mean exactly

SebGue, 2021-09-08:
Hi, I think there is some mismatch in terminology here. For non SRM the stator allways has a pole count other than the slot count. For SRM I guess the slot respectively tooth count is equal to the pole count?!

So what is the topology you actually want to design for now, i.e. how many rotor, stator slots and how many coils/phases do you want to have?

(For phase count other than 3, I would first design some similar winding with 3 phases with the generator, export the csv, edit it to my needs and import it as a user winding then.)

With 'rect' I meant the rectangular signal mentioned above.

SebGue, 2021-09-08:
... sorry, just saw that the rectangular signal function is not of numpy but scipy.

Ksp-3086, 2021-09-08:
As the SRM is outer rotor and for an automobile, stator slots-18, rotor slots-12, and phases -6 ( which I was not able to do due to the error shown earlier). I have attached the source file I got from the GUI
SRM_1_outer_rotor

SebGue, 2021-09-08:
If you are on the website for writing comments, you just need to drag your file into the comment box.

Ksp-3086, 2021-09-08:
[SRM_1_outer_rotor.zip](https://github.com/Eomys/pyleecan/files/7128298/SRM_1_outer_rotor.zip)
Here only zip file upload was possible

SebGue, 2021-09-08:
Hello, you can import the attached csv on an user winding. I don't know if this is the winding arrangement you want to achive. But if you open the csv file you will see how it works and how you can setup your own winding.
[SRM_1_outer_rotor_Winding.csv](https://github.com/Eomys/pyleecan/files/7128717/SRM_1_outer_rotor_Winding.csv)

SebGue, 2021-09-08:
I will also update the import, so you can also import fractions of winding definition later. So repeating arrangements will be easier to setup.

Ksp-3086, 2021-09-08:
So, I was going through the csv file you sent me. I understood that  in (1,2,18,6) 2 means pole pair, 18 means no of slots, 6 means- no of phases but what does 1 mean? The rad layer and tan layer mean radial and tangential layer of winding, right?

> Hi, I think there is some mismatch in terminology here. For non SRM the stator allways has a pole count other than the slot count. For SRM I guess the slot respectively tooth count is equal to the [...]

Also you mentioned a mismatch of terminology, for SRM, the number of poles on stator must be more than no of slots on rotor to prevent locking as far as I know. I read it in a research paper while I was gathering some information on them. So, how do I approach this problem to rectify the error

Ksp-3086, 2021-09-08:
[SRM_1.zip](https://github.com/Eomys/pyleecan/files/7129080/SRM_1.zip)
here is the code I have written so far

SebGue, 2021-09-09:
The tuple (1, 2, 18, 6) is defining the winding matrix size. The meaning should be (no. rad. layers, no. tan. layers, no. slots, no. phases).

I haven't run your code but it seems reasonable. I only would propose the following changes to rect_wav:
```
def rect_wav(A, time, T, tau, phi=0):
    d = tau / T
    x = zeros_like(time)
    for n in range(len(time)):
        x[n] = (A * d) + (2 * A / d) * (
            sin(pi * n * d) * cos(2 * pi * n * (time[n] + phi) / T) / n
        )
    return x
```
Atleast you have to inline 'return' with 'for'. With the 'phi' parameter you only got to define one time vector.
I also changed your indexing to 0 based as this is python standard. I hope the 1 based index was not by intention.

Ksp-3086, 2021-09-09:
Yes, the 1 based was not as I am used to MATLAB coding more so it is kind of new to me. Can you tell me what to do about the error of lamination  I am getting, as I am not able to run the simulation due to it.

SebGue, 2021-09-09:
Missed that error....
You have to set the stator polepairs = rotor polepairs, even if that maybe not valid in terms of SRM terminology.

Ksp-3086, 2021-09-09:
ok, and where do I set that I mean how? like in the library for SRM or somewhere?

BonneelP, 2021-09-09:
Hello, 

I have opened a PR related to the pole pair number for SRM: #430
I have started by removing the check on the get_pole_pair_number for SRM (and for now I return stator.winding.p)

For the definition of pole pair for SRM, would it be helpful to return p=rotor.Zs/stator.winding.qs ? Or should we return stator.winding.p ? (I'm not an expert of this kind of machine)

We also need to define the "set_pole_pair_number". For now it would set stator.winding.p=p and rotor.Zs = 2*p. As p is required to define the winding we could add a "set_pole_pair_number" method to MachineSRM that would only set stator.winding.p (and won't change rotor.Zs). 
We can also hide the p widget on the first step for SRM machines and add a p widget on the winding definition step.

What do you think would be the best solution ? I plan to work on the release tomorrow so we should be able to include these modifications.

Best regards,
Pierre

Ksp-3086, 2021-09-09:
I am also not that big of an expert but I do know these points but I think you should return the "statorwinding.p" for pole pair definition.

The p widget gets me confusing sometimes (like you can see in the above images https://github.com/Eomys/pyleecan/issues/428#issuecomment-914390938) , I do think it is better to add it on winding definition step. I am not sure how the code fully works but what does "set_pole_pair_number" refer to? Is it stator poles or the rotor poles. Both give different meanings so its a little vague for me. 

I don't think that no. of poles on stator affect Zs or slots on rotor. I have read about 12/6, 12/8 and 12/10 SRM, so I think that option is not needed.

Also, a side suggestion. If possible add an option about singly salient and doubly salient construction. For the singly salient construction no. of pole pairs*2=no of slots but in doubly salient construction no. of pole pairs= no of slots. It would be easier if this is included although you may already know that.

BonneelP, 2021-09-09:
> The p widget gets me confusing sometimes (like you can see in the above images #428 (comment)) , I do think it is better to add it on winding definition step. I am not sure how the code fully works but what does "set_pole_pair_number" refer to? Is it stator poles or the rotor poles. Both give different meanings so its a little vague for me.

For most of the machine, we have to enforce the rotor and stator pole pair number matches. This is why the option to select p is set on the "Machine Type" step. The widget on this step calls "set_pole_pair_number" which differ from one machine to another. So the simplest solution is to define that "set_pole_pair_number" for MachineSRM to only set stator.winding.p. 

We still have to set stator.winding.p since it is required by the star of slot algorithm to generate the winding pattern. 

> Also, a side suggestion. If possible add an option about singly salient and doubly salient construction. For the singly salient construction no. of pole pairs*2=no of slots but in doubly salient [...]

Can you provide us with schematics/screen shots/publications... about the machines you are trying to define/simulate (if there are not confidential) ?

Best regards,
Pierre

Ksp-3086, 2021-09-10:
Singly salient type SRM
[image]
and more widely used doubly salient
[image]

BonneelP, 2021-09-10:
Hello, 

We took the time to discussed about SRM machines and my colleague Emile found an interesting publication: J. Dong et al., "Hybrid Acoustic Noise Analysis Approach of Conventional and Mutually Coupled Switched Reluctance Motors," in IEEE Transactions on Energy Conversion, vol. 32, no. 3, pp. 1042-1051, Sept. 2017, doi: 10.1109/TEC.2017.2672741.
https://ieeexplore.ieee.org/document/7862226

So here is our proposal to handle SRM machines: 
- p makes no sense for SRM machines so this parameter will be hidden in the GUI "Machine Type" step (for SRM). The winding pole pair number will be calculated automatically for the call to star of slots method and will be stored in winding.p only for compatibility, since its value is meaningless and is not used afterwards. Then we can hide the value on the "Stator Winding" step as well (for SRM). According to the publication we should have something like Zs=qs*p
- In the simulation we use "p" to define the current frequency as f=N0*p/60. Then for a SRM machine, we plan to return "p=Zr" to have f = N0*Zr/60
- Regarding Singly /doubly salient, the difference seems to be the number of layer of the winding (1 "distributed winding" or 2 "tooth winding"). The winding pattern can also be tuned with the coil pitch that will be displayed and editable in the GUI (we need to work on the coil pitch GUI).

I will implement these modifications in the dedicated PR. What do you think about this proposal ? 

Best regards,
Pierre

Ksp-3086, 2021-09-10:
I think it is great and Zs=qs*p makes more sense for an SRM.

You referred current frequency in SRM? Is the current going to be AC? Correct me if I am wrong but SRM does use a DC Current with a switching behaviour. If you are referring to switching as frequency then that seems fine.

BonneelP, 2021-09-10:
Hello,

I just release pyleecan 1.3.3 that should help you going forward with SRM machines (the GUI doesn't talk any more about p for SRM). You can use the command "pip install -U pyleecan" to get this new version. 

Best regards,
Pierre

Ksp-3086, 2021-09-10:
[image]
looks like it is not being upgraded or installed. I have uninstalled and reinstalled again but I get the same error. I will try again tomorrow as its 11 pm here now.

EmileDvs, 2021-09-12:
> 
> 
> I think it is great and Zs=qs*p makes more sense for an SRM.
> 
> You referred current frequency in SRM? Is the current going to be AC? Correct me if I am wrong but SRM does use a DC Current with a switching behaviour. If you are referring to switching as frequency then that seems fine.

Hi Ksp,

Yes exactly the "current frequency" means how many times each phase is fed per rotor revolution. 

Concerning stator winding supply, it is a good start indeed to enforce step wise currents and check magnetic results. I guess that in fact the SRM is more easily controlled by step wise voltage instead of currents, and the resulting current would be smoother due to high frequency filtering from stator winding inductance. 

For your information, non-linear inductance calculation function of rotor position can be implemented in pyleecan for SRM without too many efforts since it has already been done for induction machines and synchronous machines. Don't hesitate to open an issue if you are interested in this.

Best regards,

Emile

Ksp-3086, 2021-09-12:
> Hi Ksp,
>
> Yes exactly the "current frequency" means how many times each phase is fed per rotor revolution.
>
> Concerning stator winding supply, it is a good start indeed to enforce step wise currents [...]

Hi Emile,
I am just getting started with SRM so I too do not know much about how to implement the non-linear inductance calculation. I am actually working on this as I was asked to calculate an efficiency map as an internship. The motor is my own design and although I have tried other softwares, pyleecan is the closest I have gotten so far to achieving my goal. Although the inductance calculation is a but complex for me, I would love to know more about it. Could there be a documentation you would recommend? Right now, I am waiting for the installation error to be fixed so that I can start working again. 

Also, I read through the tutorial for Toyota Prius IPMSM machine. The currents were AC type so I could not  get the non-linear inductance calculation. Is that the same tutorial code you are referring to or the another one?

Ksp-3086, 2021-09-14:
Hi @BonneelP ,

I do not mean to intrude since I know that you are working on updates on pyleecan. I just wanted to ask if you had a look at the issue for the upgrade to version 1.3.3. Maybe you might have fixed that but I do not know it is still causing problems for install. If the issue is fixed then I think that it may be causing problems due to cached pywin32 which was downloaded before or other stuff. How do I remove that cached versions?

I have already used the command "pip --no-cache-dir install" and "pip cache dir "

BonneelP, 2021-09-14:
Hello, 

On my side I have no problem when installing the latest version of pyleecan from pip :/ 
Maybe you can try to update the version of pip as suggested in the error message ? If the error is linked to pywin32 you can also try to uninstall it before installing pyleecan. Do you use other packages that would require this dependency ?

Best regards,
Pierre

Ksp-3086, 2021-09-14:
[image]

Hi @BonneelP , 
I installed the new version of pip. It did go beyond pywin32 but the backtracking for like every version is taking a lot of time. Then after it checks pluggy, it stops for an hour and gives the error. I do not know why is this happening and pyleecan is the only package I am using from python. My python version is 3.8.10. Any workarounds you can suggest as I could not find much help on google

BonneelP, 2021-09-14:
Hello, 

I think you are experiencing an issue linked to the new way pip handles dependencies (cf. error message bellow) I tried to install pyleecan on two computers on it works on both. Although, on my second try I got the following message:
[image]
Our requirements doesn't seems to be consistent any more, we are planning to do a release of SciDataTool in the near future, we already have corrected the requirements. Then after a new release of pyleecan (with only updated requirements), hopefully this may solve your problem. On a side note, on my second computer I'm using Python 3.8.6 (installed from python website) and pip 20.2.1

Best regards,
Pierre

Ksp-3086, 2021-09-15:
Looks  like pip version 20.2.1 helped in install. I made the whole model from scratch again.  I got a new error saying Lamslot has no attribute Zs. Any suggestions on this?

[image]

BonneelP, 2021-09-15:
Great :) 
Indeed LamSlot don't have "Zs", it should be "get_Zs()". In your screenshot I can't see in which file the error is occurring. Can you further investigate to see where the bug is coming from ? 
Best regards,
Pierre

Ksp-3086, 2021-09-15:
It doesn't specify where exactly it comes from just the code that I am typing. In the terminal it backtracks only to the python code file or the installed directory and the vs code extension for python. You can see I have attached the code and the screenshot of the terminal if it helps

[image]
[SRM_1.zip](https://github.com/Eomys/pyleecan/files/7168645/SRM_1.zip)

BonneelP, 2021-09-15:
Thanks, I will correct it as soon as possible and create a new release (it may be this evening)

BonneelP, 2021-09-15:
Hello,
I have release pyleecan 1.3.4 and I managed to run a FEMM simulation with a SRM. The next step would be to make sure that the simulation makes sense ^^" Do you have any validation case / publication that we can use ? 
In particular I'm wondering what should be returned by LamSlot.comp_angle_d_axis (for now it returns 0)

Best regards,
Pierre

SebGue, 2021-09-16:
Hello Pierre,

in my opinion it should eighter be the angle (of the middle) of the first tooth or the first slot. But I would rather go for the middle of the first tooth. The reason is that this seems the common d-axis definition in reluctance rotors and LamSlot is some sort of reluctance rotor.
(This may seem to be contradictory to PM reluctance rotors (and also IPM rotor), but will make some sense if you think of the pole as a tooth or the tooth as a pole :-) )

Best regards, Sebastian

Ksp-3086, 2021-09-16:
Hello @BonneelP ,

Before modelling in pyleecan, I was trying my hand on ANSYS Maxwell. The drawback was that it did not support outer rotor for SRM, and the only documentation I have for SRM are from ANSYS Maxwell. I could share links with that with you. These documentations show flux density and torque graphs with respect to rpm and electronic degree.
http://ijceng.com/gallery/24-cej-2929.f.pdf
https://www.researchgate.net/publication/320662784_Modelling_and_simulation_of_SRM_based_automatic_transmission_system_for_hybrid_vehicles_in_Ansys_Maxwell
https://www.researchgate.net/publication/343340518_Understanding_Switched_Reluctance_Motor_Analysis_using_ANSYSMaxwell
https://www.researchgate.net/publication/322347406_Switched_Reluctance_Motor_Initial_Design_for_Electric_Vehicle_using_RMxprt
I am not fully sure about what to return for LamSlot.comp_d_axis as I am not that good with SRM yet.

I do have some extra questions though
1.  Is it possible to reduce time taken by solver, I saw it taking 1e-08 steps, it took 2 hours for the simulation to run (and the reason I was not able to reply early)
2. I know for torque graph we have "out_femm.mag.Tem.plot_2D_Data("time")" command but what is the command for calculating losses?
3. I need to out output speed(rpm),Torque, losses in a dat file (so that i can also run the files in MATLAB so any other format that could run this could also suffice)  so that I can share the data with my colleagues, since I am new to pyleecan and python at the same time, how do I do that?

Ksp-3086, 2021-09-16:
Hi,
I think I may have done something really wrong. I simulated the current waveforms to be like this, different colours show different phases(I cannot get to plot them in subplots in python yet)
[image]
The torque is unbelievably high as you can see
[image]

SebGue, 2021-09-16:
Hello Ksp,

if the torque is high depends on the size of your machine. If you think its too high, yoiu may check the current density for reasonable values. Futher, do you use linear or nonlinear BH curves? Linear ones may also give unreasonable high torque values.
Regarding the calculation time, you should only simulate the least needed periodicity. Depending on your machine this may be even only one current pulse.
For the losses, I would only care about them, if everything is okay with your simulation. Otherwise it wont make any sense.
BR S

Ksp-3086, 2021-09-16:
Hi @SebGue ,
I used the material B-H curve. The size of motor is that (as it is outer rotor), the Rext of rotor is 200mm. Can you be clear on what you mean by simulate the least needed periodicity? 

On the losses, I have not been able to find how to call and get the values. I know iron losses are possible but how? I have attached the code if you want to take a look at it. I think the SRM Tutorial may almost be done except for a proper writing of the code once its able to run
[SRM_1.zip](https://github.com/Eomys/pyleecan/files/7178881/SRM_1.zip)

SebGue, 2021-09-16:
Looking at you winding setup (even though your slot fill is low) the current density is too high at 100 A. You got to check this. For 'normal' machines (depending on cooling type) the effective continuous current density is maybe in the range of 3 A/mm² to 10 A/mm². Since yours is way higher, your torque is also too high.

Your current wave form as well as the torque curve is repeating within your choosen simulation time, so you got some periodicity within. If you have no reason for doing so, you should reduce the simulation time (i.e. the time vector) to the least needed time period. This is what I meant with least needed periodicity.

For the losses you could have a look at the unit test. I will look for the file later, if you don't find it yourself.

SebGue, 2021-09-17:
Hello, you can find an example of loss calculation in 'Tests\Validation\Loss\test_FEMM_loss.py'. 
The loss calculation is maybe not intuitive to use. So if you have some ideas on how to improve, you can join the discussion in #59.

Ksp-3086, 2021-09-17:
Hi,
I looked at the file which you mentioned, I have a few questions

1.  As of now, is only iron loss supported? I mean copper losses (I^2*Rt) maybe easy to calculate but I do not know if it is there.
2. I pulled this file (tests folders) from github but it is showing an error saying that module tests is not found, Do I need to transfer it somewhere else like in the python directory?
3. How do I like actually call it? being new to python I am still figuring things out. Just an example would suffice

EmileDvs, 2021-09-17:
> Hello Pierre,
>
> in my opinion it should eighter be the angle (of the middle) of the first tooth or the first slot. But I would rather go for the middle of the first tooth. The reason is that this [...]

Hi all, 

I agree with @SebGue on rotor initial position, the most common term in the literature seems to be "aligned position" (respectively "unaligned position"), meaning one rotor tooth middle in front of one stator slot middle (respectively one rotor tooth middle in front of one stator tooth middle). 

In theory positive torque is created between "unaligned" and "aligned" position so the phase current in the stator tooth should be non zero only between both rotor position, while current in other phases is zero (as you show in your previous current graph).

In the torque graph you showed, there are some negative peaks of torque that decrease your average torque and increase torque ripple. This is msot likely because of current still flowing in the stator tooth winding while the rotor tooth has already passed aligned position. If you impose square current wave with proper rotor initial position I think you should manage to get low torque ripple.

Emile

SebGue, 2021-09-17:
Hi @Ksp-3086 ,

there are also I²R losses (LossModelWinding) in the test file.
For running the file you would need to do some cleanup (e.g. remove from Tests...), but I think you don't need to run it.
You should only adapt the 'Setup the Loss Models' part. Make sure you use `simu.mag.is_get_meshsolution = True` to store the FEA solution.
To get the results, have a look at the 'Run the Loss Simulation' section.

Best regards, Sebastian

Ksp-3086, 2021-09-17:
Hi @SebGue ,

So, I can just copy the code and use it, right? I thought I needed to call the function defined. I will try and implement it as much as I can.

Ksp-3086, 2021-09-17:
Hi @EmileDvs ,
As I was not able to find a proper square or rectangular function (the one is python takes inputs only as time and duty cycle, so the switching frequency was same no matter what the duty cycle), I made my own using for and if statements. Presently the pulse width (modelled according to rpm) is  60/(stator slots*N). Do you think I should decrease the pulse width? The time period between switching on-off the same phase is phases*pulse width? Any suggestions if I should take another approach or modify this somehow to decrease torque ripple?

BonneelP, 2021-09-17:
>     I need to out output speed(rpm),Torque, losses in a dat file (so that i can also run the files in MATLAB so any other format that could run this could also suffice) so that I can share the data with my colleagues, since I am new to pyleecan and python at the same time, how do I do that?

There is a function to save directly in ".mat" which should be easier to use in Matlab: https://docs.scipy.org/doc/scipy/reference/generated/scipy.io.savemat.html

Ksp-3086, 2021-09-17:
Hi all,
I saw this code "myIronLoss.N0 = [4000, 6000]  # list of speed to override actual speed" in  the file for test_loss, why is the actual speed overridden? Also can I run output the normal losses from the output along with the normal simulation?

`out=Output(simu=simu_SRM)
out_femm = simu_SRM.run()

loss = out.loss
mshsol = loss.get_loss_dist(part_label="Stator", index=0)`

I did not know which was unnecessary so I kept both output and out_femm.

SebGue, 2021-09-17:
Hi @Ksp-3086 

the function is an unit test, to test if one get valid results if there are code changes.
The list of speeds is just to calculate losses for multiple speeds, without the need to recalculate the whole magnetic simulation for each speed. This is possible due to the fact that the fea is static.
Regarding current wave form, I think there are multiple technique e.g. some current ramp up to reduce torque ripple. There should be tons of literature in that topic :-)
For the different outputs, since python is object oriented, you have to think of references to an obkect. So you don't store multiple copies of the output in your code. You only store the reference to the object to multiple variables.
Hope that helps.

Ksp-3086, 2021-09-19:
Hi,
I am trying to save the Torque values from this\
`T=out_femm.mag.Tem` but I get "ValueError: Expected 1D or 2D array, got 0D array instead" in the terminal. Further more even if I try to use the values of T in the code, I get the error "Datatime object not iterable"

[image]

I am also getting high losses, I think I might need to decrease the number of turns for that and the low mechanical power may be due to high torque ripple which causes low average torque

[image]

SebGue, 2021-09-21:
Hello, 
don't know how you tried to save the torque. But if there is an issue with the save method, you'd post this bug to SciDataTool repo.
For your first figure, you use some statistics package?! This is unlikely incompatible with SciDataTool. So you can eigther get the ndarray from the torque Data object with its dedicated methods and calculate the mean value after that. Or you use the Data objects slicing capabilites (see SciDataTool slicing tutorial) that also include averaging.
If you think the losses are calculated are wrong, you could verify them with some 'pen and paper' calculation easily. Actually there is only one test case so there may still be bugs in the calculation.
Nt_tot and Na_tot are properties of InputCurrent that are time discretization and angular discretization (i.e. number of sampling points of the field along the airgap). Since Na_tot is not dividable by 3, the whole machine is simulated instead of only 1/3 of the machine. 
BR S

Ksp-3086, 2021-09-21:
Hi,
For the stator winding losses, I guess the formula must be I^2R.  I gave the value of I myself but how do I get the value of R, I mean yes it is possible from the internet but there are other parameters for eg- length, area and changes according to temperature. For the Iron losses, I am not so sure how to calculate them. I need some parameters and then I can check, how do I get those?

[37 more replies]

closed 2021-10-04 by Ksp-3086