Eomys/pyleecan #433: How to reset rotor initial position in the simulation

Sijie-NI, 2021-09-16:
Hello, 

After modeling my PMSM in Pyleecan, I simulate this machine with FEMM. Normally, the torque should reach the maximum,  if I impose Id =0 and Iq=Imax, but in fact, it is always around zero. If I inverse Id and Iq, Id=Imax, Iq =0, the torque can attain the maximum. After the reflection, I think Pyleecan requires the current vector is aligned with the rotor d axis at the initial condition (t=0). But in my case, the current vector is aligned with the q axis. To solve this problem, I have to rotate my rotor of  22.5 degrees. Could you tell me how to reset the rotor position in the simulation or rebuild this model? 
[image]

EmileDvs, 2021-09-17:
Hello Sijie,

I think this may be because d-axis angle is not accounted for for spoked rotor topology. In spoked rotor the pole is in between two magnets due to flux concentration yielding a d-axis in between two NS magnets and a q-axis in front of magnets.  The current convention for IPMSM and SIPMSM (independently of spoked magnets) is exactly the opposite:  d-axis in front of magnet, q axis in between magnets. 

Therefore it is reassuring that you get physical torque results if you swap Id and Iq :) You'll probably have to add this specific rotor spoked case in comp_d_axis_angle method of LamSlotMag lamination to get proper torque for Id and Iq.

Best regards,
Emile

Sijie-NI, 2021-09-17:
Thanks, Emile, 

You are right. I check the actual comp_d_axis_angle that sets always the d-axis angle as pi/p/2. That is the reason why Pyleecan confuses the d-axis and q-axis. I can add this case in LamSlotMag by creating an "if" statement for example. But could you tell me with which condition (argument in the "if" statement) I can use to distinguish the normal IPMSM and this special IPMSM?

Best regards

Sijie

SebGue, 2021-09-17:
Hello Sijie,

isn't there a specifc 'Hole' for that type of geometrie? So what's your actual hole and how do you manage the magnetization direction?

Best regards, Sebastian

Sijie-NI, 2021-09-17:
Hello Sebastian, 

Yes I create a specfic Hole "HoleMLSRPM" in Pyleecan. Because of the radial magnets. The magnetic field is special. The north pole and south pole are in the middle of magnets
[image]

Best regards,

Sijie

BonneelP, 2021-09-17:
Hello,

So the needed modification would be:

```python
LamSlotMag.comp_angle_d_axis: 
if self.magnet.type_magnetization == 3:
     return 0
else:
    return pi/p/2
```

Is it correct ? 

Best regards,
Pierre

Sijie-NI, 2021-09-17:
Hello, Pierre, 

Yes correct, with the help of Emile, I have programed the similaire code in LamHole.comp_angle_d_axis: 

    p = self.get_pole_pair_number()
    if self.has_magnet() and self.hole[0].magnet_0.type_magnetization!=3: #Normale rotor with magnets
        return pi / p / 2
    else: #Without magnet or spoke rotor with magnets
        return 0
In your opinion, should I change LamSlotMag also?  

Best regards, 

Sijie

BonneelP, 2021-09-17:
Hello Sijie,

Yes I forgot that you were working on an IPMSM. So we should correct both. 

Best regards,
Pierre

Sijie-NI, 2021-09-20:
Hello, I have pushed my modification about LamHole and LamSlotMag in the branch LSRPM. 

Thanks for your help,

Best regards
Sijie

closed 2021-12-09 by EmileDvs