Eomys/pyleecan #434: [CO] Always display coil pitch in winding GUI + assign default coil pitch value if 0 or None

EmileDvs, 2021-09-17:
Hello, 

I noticed some issues with machines whose winding coil pitch is not initialized, i.e coil pitch is None or 0 (depending on pyleecan version I presume). I propose to initialize undefined coil pitch in function of slot per pole and per phase (spp) value such as:

- if ssp >1 (distributed winding) : coil_pitch is qs*spp
- if spp <1 (concentrated winding) : coil_pitch is 1

This initialization is made both in comp_connection_mat of Winding object and in SWinding GUI for consistency.

Besides, I also propose to always display coil pitch parameter in the GUI, even for single layer winding. In fact it can be relevant to modify coil pitch for single layer winding. For example this parameter enables to choose between tooth concentrated and distributed winding, both being physical for some topologies and wishable depending on the application.

These modifications will be available in a future PR, in which I'll refer to the present issue.

Please let me know if you have some remarks/questions about this.

Best regards,
Emile

SebGue, 2021-09-17:
Hello Emile,

👍 for your proposal. Instead of spp = 1 as the decision criteria, I would recommand > 0.5 since there is e.g. distributed spp = 3/4 windings.

Lately I forgot to set coil_pitch to the correct value in WindingUD and got some wrong end-length calculated. So it would also be great, if the coil pitch is checked in the GUI and give some warning if there is too much deviation from qs*spp.

Best regards, Sebastian

SebGue, 2021-09-17:
... further, it's not possible to set coil_pitch in the GUI for WindingUD

EmileDvs, 2021-09-17:
Thanks for your answer @SebGue, I modify default value to 0.5.

For WindingUD relying on csv import, coil pitch is forced by imported matrix. Is there a formulae that enable to caculate coil pitch properly, assign it to the object and display it as label on the GUI (non editable) ? 

I'm thinking of something like calculating the number of slots between Nturns and -Nturns on the first phase but I don't know if it is generic and robust at all.

SebGue, 2021-09-20:
Hello Emile, 

I'm not sure if there is a formulae to calculate coil pitch. Further more for one connection matrix there (may) exist multipe ways how to actually connect the conductor. Let me know if you need some example of what I mean.

For the calculation we could also utilize swat-em. But I'm not quit sure, since it sometimes outputs multiple coil spans. Maybe for the above reason.

I thought that the WindingUD's coil_pitch parameter should reflect the effective coil pitch (on the acutal 'in production machine') used for the end winding length calculation. 

👍 for the warning. If we base it on 'qs*ssp' there should be a ssp dependent tolerance (e.g. ssp = 2 -> +- 1, ssp = 4 +- 2, ... ). 

Best regards, Sebastian

BonneelP, 2021-09-24:
Done in #437

closed 2021-09-24 by BonneelP