Eomys/pyleecan #447: [BC] GUI + Slot10.H1

SebGue, 2021-10-03:
Hello,

there is an issue with Slot10 H1 property in the GUI. It is maybe related to the conversion to 'mm'.
Each time one input a new value, go to the next step and come back to slot page, H1 gets smaller by 1e3.

I would really appriciate if someone can fix this issue since I may not have the time to do so. :-)

Best regards, Sebastian

BonneelP, 2021-10-04:
Hello,

It should be solved in #450. As the test worked, it may have been a display error. It seems that it was only for SlotW10 the other that uses the "h1_is_rad" property already handle [mm] properly.
Thank you for spotting this one :)

Best regards,
Pierre

BonneelP, 2022-09-06:
It was indeed corrected in #450

closed 2022-09-06 by BonneelP