Eomys/pyleecan #476: [BC] Optimization tutorial doesn't work

HugoKulesza, 2021-11-15:
I am trying to use the tutorials to learn how to use Pyleecan. When using the tutorial for optimization (tuto_Optimization.ipynb), the call to solve() does not work.
```
from pyleecan.Classes.OptiGenAlgNsga2Deap import OptiGenAlgNsga2Deap
solver = OptiGenAlgNsga2Deap(problem=my_prob, size_pop=16, nb_gen=8, p_mutate=0.5)
res = solver.solve()
```

When it is executed, many instances of FEMM are opened but remain empty and don't close when the execution end.
For every generation and simulation launched by the resolution, the following error appear : 

```
Design Variables: SW0: 4.72e-03, Rext: 8.01e-02
[14:25:16] Starting running simulation EM_SIPMSM_AL_001 (machine=Toyota_Prius)
[14:25:16] Starting Magnetic module
[14:25:18] The following simulation failed :
Traceback (most recent call last):
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Functions\Optimization\evaluate.py", line 46, in evaluate
    indiv.output.simu.run()
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Methods\Simulation\Simulation\run.py", line 52, in run
    self.run_single()
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Methods\Simulation\Simu1\run_single.py", line 31, in run_single
    self.mag.run()
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Methods\Simulation\Magnetics\run.py", line 24, in run
    out_dict = self.comp_flux_airgap(output, axes_dict)
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Methods\Simulation\MagFEMM\comp_flux_airgap.py", line 97, in comp_flux_airgap
    FEMM_dict = draw_FEMM(
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Functions\FEMM\draw_FEMM.py", line 150, in draw_FEMM
    surf_list.extend(machine.rotor.build_geometry(sym=sym))
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Methods\Machine\LamHole\build_geometry.py", line 45, in build_geometry
    transform_hole_surf(
  File "C:\Users\hkule\AppData\Local\Programs\Python\Python38\lib\site-packages\pyleecan\Functions\Geometry\transform_hole_surf.py", line 36, in transform_hole_surf
    assert Zh % sym == 0
AssertionError
```

BonneelP, 2021-11-26:
Hello,

Thank you for noticing. I'm currently updating all the tutorials to match the latest modifications in pyleecan. I will release pyleecan 1.3.7 once all the tutorial are up to date. 

Best regards,
Pierre

BonneelP, 2021-11-26:
Pyleecan 1.3.7 is now available, can you try again with this new version ? I have updated the Optimization tutorial and it worked on my side. Not all tutorials are up to date, I will update the website once it will be the case.

HugoKulesza, 2021-11-29:
It's now working well on my side too !
Thank you for your updates, it will help.

Cordially,
Hugo

closed 2021-11-29 by HugoKulesza