Eomys/pyleecan #478: Problems Importing DXF file with GUI

CandleAfro888, 2021-11-22:
Hello. I'm just a beginner using Pyleecan. My version is:

[image]

Nevertheless, i was trying to use the DXF import utility within the GUI when defining the rotor holes and the following dialog window shown up.

[image]

Looking at the Jupyter notebook cell it shows no error tracing. So I'm not able to see where is the problem. 

¿Can someone help me?

Thanks

BonneelP, 2021-11-26:
Hello, 

Sorry for the late answer, I was in holiday. On the current master of pyleecan a more recent version of the DXF import is available. I just checked and I managed to import a hole with it. I will soon release a new version of pyleecan with these modifications.

Best regards,
Pierre

BonneelP, 2021-11-26:
Pyleecan 1.3.7 is now available, can you try again and confirm if your issue is corrected ?

tobsen2code, 2022-03-30:
Hello together, 
I am working with Pyleecan version 1.3.8. and getting the same error. 
Additional I tried to import and export the dxf data with FEMM without any improvement. 
Do you have any solution for this problem? 
Thank you for your support.

Best regards,
Tobias

BonneelP, 2022-09-06:
Hello,

The import DXF has been rework with PR #485  and pyleecan v1.4.0 was released and I managed to use DXF import with the exe version. 
Can we close this issue or do you still have errors on your side ? 
Best regards,
Pierre