Eomys/pyleecan #488: Install in conda env

gforti84, 2022-01-17:
Hello, tried installing using conda environment, but can't run the GUI.
Created the conda env (python = 3.8.12) and installed pyleecan using pip install. 
[image]

gforti84, 2022-01-17:
Solved downgrading 'pywin32' version 
pip install pywin32==228

https://stackoverflow.com/questions/58612306/how-to-fix-importerror-dll-load-failed-while-importing-win32api

gforti84, 2022-01-18:
Another issue appeared. Trying the tuto_MeshSolution tutorial. I saw that the folder 'Tests' in the ROOT_DIR folder did not came with the 'DATA' folder. I installed using 'pip install pyleecan' within a conda env. I copyied it from the git clone folder.

BonneelP, 2022-01-18:
Hello, 
Thank you for your returns. We plan to work on the installation procedure and on the website once the next version of the software will be available . We will include both your modifications then. 
Best regards,
Pierre

BonneelP, 2022-09-06:
Hello,
The installation page is updated and the requirements are now handled with optional package.
Can we close this issue or do you need further help on the subject ? 
Best regards,
Pierre