Eomys/pyleecan #495: Help with loss model implementation

Anshrajdev101, 2022-02-23:
Hello, 
I have been trying to implement the loss model for Bertotti using PYLEECAN. Unfortunately, I am receiving the following error which I am not able to resolve. It would be a great help if someone could help me understand what the problem is. 
[image]
My PYLEECAN version is given below.
[image]
Kind Regards
Ansh

SebGue, 2022-02-23:
Hello Ansh,

I guess I've got to update the loss model due to recent (internal) changes in pyleecan objects.
For now, I think you should use your own Bertotti model coefficients to get some results.

Best regards,
Sebastian

Anshrajdev101, 2022-02-23:
Hello Sebastian,
Thank you for your quick response. I have added my own Bertotti model coefficients and I am getting the desired results. I have another doubt in relation to the loss models. I am trying to implement a lumped parameter thermal network that requires me to have the stator teeth and stator yoke iron losses calculated separately. Do you have a suggestion regarding the implementation of this?
Kind Regards
Ansh

SebGue, 2022-02-24:
Hello Ansh,

indeed the separation of losses is some required feature. 
You could have a look at LossModelBertotti\comp_loss (starting at line 112) and adapt the code. I.e. I would compute the cell areas, set the area of some cells to zero (e.g. all cells that are in the yoke) and calculate the losses.
If you need any futher help on this, just ask.
Otherwise it would be great if you could provide at least a script with that feature, so we can implement it as a dedicated method.

BTW: Do you plan to contribute the thermal model?

Best regards, Sebastian

Anshrajdev101, 2022-02-28:
Hi Sebastian,
Sorry for the late reply. On implementing the code for the Bertotti loss model, I found that the code uses FFT to calculate the losses, however, I wanted to calculate the losses from static FEA and thus would be using a different approach. I can provide you with that code if you want.
Regarding the thermal model, the code I am writing is applicable for Surface PMSM only. I can provide you with my script and a lot of it can be reused for other machines however I am unsure if I'll be able to include it in PYLEECAN's architecture(I am a beginner with python and still figuring it out).
Kind Regards,
Ansh

SebGue, 2022-02-28:
Hello Ansh,

don't worry, I'm usually even later with my replies.
I'm very interessted in your static FEA approach. I also got a single FEA loss estimation that is not integrated in public pyleecan yet. So I'm looking forward how it compares to yours.
Thermal modeling is also of interest, so you could provide your script and I will see how it can be integrated. Here I also got some simple FEA approach to compare results.

Best regards, Sebastian