Eomys/pyleecan #530: Help about Winding classes

cjw1254, 2022-04-15:
Hello. It's my first time using the pyleecan so I'm following the tutorials first. (Tutorials: https://github.com/Eomys/pyleecan/blob/master/Tutorials/tuto_Machine.ipynb)

While I m following the first tutorial, the result is the same as answer
[image]

Also, there r another issue on 2nd tutorial as follows, and I think there r no Winding classes for now.

[image]
[image]

I ve tried another winding class 'WindingCW2LR', but error was the same as above.

Is it a problem with version?? or can I add winding classes by myself??

I'm using python ver. 3.7.13 with anaconda & VS code
Also Pyleecan info is as follow:
Running date: April 15, 2022
Pyleecan version:1.3.8
SciDataTool version:2.3.3

Thx a lot for superb SW and I hope I can contribute someday.

BonneelP, 2022-04-15:
Hello, 
Thank you for your interest in pyleecan.
Sadly the tutorial are not up to date :/ I should have time to work on that on next week.
By waiting WindingCW2LT doesn't exist any more, now there is only one class named "Winding" where you can set coil_pitch, qs, p, Nlayer and the corresponding optimal connection matrix is computed thanks to star of slot method from swat-em. Normally all that can be done from the GUI. 

Best regards,
Pierre

cjw1254, 2022-04-16:
Oh I see, you mean the class "Winding" documentated as follow?
: https://www.pyleecan.org/pyleecan.Classes.Winding.html
I m trying to work on it with language not GUI so I ll follow the documentation above first.
If it allows ur time, plz let me know when the tutorials done.
Appreciate ur kind reply.

BonneelP, 2022-04-19:
Yes it's this class. You can also find the csv description here https://github.com/Eomys/pyleecan/blob/master/pyleecan/Generator/ClassesRef/Machine/Winding.csv 
You can also take inspiration from the machine library: https://github.com/Eomys/pyleecan/tree/master/pyleecan/Data/Machine
open any machine json and search for "winding", for instance Prius: 
[image]
[image]

Best regards,
Pierre

BonneelP, 2022-09-06:
Hello,
It took me longer that expected but the tutorials are now up to date, same for the website. 
Do you need any further assistance or can we close this issue ?
Best regards,
Pierre