Eomys/pyleecan #531: plot_schematics partially ignores fig and ax parameters, and produces an extra figure.

mjfwest, 2022-04-20:
```python
figs = list(map(plt.figure, plt.get_fignums()))
print('Number of figures {0}'.format(len(figs)))

# create two side by side subplots on one figure
fig22, (ax1, ax2) = plt.subplots(1, 2)
# plot the slot shape on the left hand subplot
stator.slot.plot(fig=fig22, ax=ax1)
# try and plot the schematics on the right hand axis
stator.slot.plot_schematics(fig=fig22, ax=ax2, is_default=False,
                            is_show_fig=False,
                            is_add_point_label=True)
# Two things have gone wrong already, 1. plot_schematics has taken fig and ax
# as parameters, but it seems to ignore the ax parameter when plotting the slot
# shape but uses the correct ax when plotting the labels.  2. it has created
# another figure that I didn't ask for with the stator and winding plotted.

figs = list(map(plt.figure, plt.get_fignums()))
print('Number of figures {0}, and fig22 is one of them {1}'.format(
    len(figs), fig22 in figs))
fig22.savefig('error_plot_part_2.png')
```

[image]
[image]

The culprit for the first part is probably this section of `plot_schematics.py:
```python
        # Getting the main plot
        if self.parent is None:
            raise ParentMissingError("Error: The slot is not inside a Lamination")
        lam = self.parent
        lam.plot(alpha=pi / self.Zs, is_show_fig=False)  # center slot on Ox axis
        if fig is None or ax is None:
            fig = plt.gcf()
            ax = plt.gca()
```
Changing it to `if fig is None and ax is None:` will fix this problem, but will need checks for `fig is None and ax is not None:` etc.

BonneelP, 2022-04-22:
Hello,

Which slot are you using ? I haven't checked all the slots but it seems that most of them share the following arguments:
[image]
In most of them, we call a plot of the lamination then get the corresponding fig and ax to add the schematics information.
[image]
Maybe we could do better by calling the init_fig function in the plot_schematics method and passing fig, ax to lam.plot().

Do you need the fig, ax argument to add the schematics on an existing plot ?

Best regards,
Pierre

mjfwest, 2022-04-26:
This is using `SlotW23`.  In general yes, I want to control where the plot ends up, and what it looks like  The really cool bit that pyleecan is doing for me is generating all the patches, annotations, etc, and adding them to an axis; every assumption it makes after adds little benefit, and often needs correcting.  

I think your suggestion sounds like a solution.  Does the `init_fig` method exist, or does it need writing?

BonneelP, 2022-05-02:
cf final answer to #527

closed 2022-05-02 by BonneelP