Eomys/pyleecan #541: Error collecting test session

Theumazz, 2022-04-29:
Hello everyone, when I try to install Pyleecan I get following error
[image]

I installed all required packages. I use Anaconda and Spyder. Does someone know how I can solve this error?

With kind regards,

Thomas

SebGue, 2022-04-30:
Hello Thomas,

I guess you followed the installation instruction on the website that need some update.
Futher I guess you tried to run the test, i.e. `python -m pytest -m` which is not necessary. 
You can just run pyleecan e.g. with `python -m pyleecan`.
If you still want to run the tests, it seems you need to grant access to the AppData folder, i.e. grant administrator rights to your console.

Best regards, Sebastian

Theumazz, 2022-05-14:
Thanks a lot Sebastian, it worked with `python -m pyleecan`

BonneelP, 2022-09-06:
Hello,
I have updated the installation guide, for "pip install pyleecan" method the tests are not mentioned anymore and "python -m pyleecan" is. Thank you for reporting that. 
I closing this issue, you can reopen it if you need further assistance.
Best regards,
Pierre