Eomys/pyleecan #542: Installation issue

fixundfoxi1, 2022-05-03:
Hi,
while installing pyleecan most of the installation process runs through however it stops with following message:
  Using cached deap-1.3.1.tar.gz (1.1 MB)
  Preparing metadata (setup.py) ... error
  **error: subprocess-exited-with-error**

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in deap setup command: use_2to3 is invalid.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Any idea?
Thanks for your help!

BonneelP, 2022-05-03:
Hello,

Thank you for your interest in pyleecan.

If I understand correctly, it seems that pip doesn't manage to install deap 1.3.1. 
As a first solution, you can try to run directly "pip install deap" with administrator rights. 1.3.1 is the latest version and I'm using it on a Python 3.8 so there should be no reason why it can't be installed. Which version of python are you using ?
Once deap installed you can run "pip install pyleecan" again.

As a further response to your issue, I'm experimenting "optional" dependency. deap is in fact not mandatory any more in pyleecan. This package enables to run NSGA2 optimization, we now have Bayes optimization with smoot as well. So as a test I have removed deap and smoot from the mandatory dependencies from pyleecan (to enable to choose which package to use).
They are listed in the "full" list, 
`pip install pyleecan`
would install the smallest number of package and 
`pip install pyleecan[full]`
 the complete list (normal + deap + smoot + gmsh). 

So technically, the "pip install deap" from the first solution could not be required if you don't plan to use NSGA2 optimization.

@SebGue What do you think of this new organization (I'm taking inspiration from what we did for https://github.com/Eomys/MoSQITo) ? It's an experimentation before 1.4.0. By the way I'm currently working on updating all the tutorials, then I plan to update the website and the "how to install page".

Best regards,
Pierre

BonneelP, 2022-05-05:
Hello,

Glad to know that you managed to install pyleecan :) We plan to move to upper python version in the near future.
I haven't seen this message before :/ Are you using windows or Linux ? We still have to work on the compatibility with Linux cf issue #104 (hopefully there was an update that should enable to do it now)

Best regards,
Pierre

SebGue, 2022-05-12:
Hello @BonneelP 

sorry for my late response. Do you mean optional dependencies by new organisation? That seems reasonable.
Or is there anything else new to mosquito, since I don't follow this project.

Best regards, Sebastian

BonneelP, 2022-05-12:
Yes I was referring to the optional dependencies.

BonneelP, 2022-09-06:
Hello,
I just released pyleecan 1.4.0 that includes the optional dependencies installation and I have updated the website.
I'm closing this issue, if you need further help you can reopen it.
Best regards,
Pierre

closed 2022-09-06 by BonneelP