Eomys/pyleecan #544: InputCurrent error when running magnetic module from tutorial

Theumazz, 2022-05-14:
Dear sir, dear madam,

I am following along the webinar/tutorial with a machine I made myself in the GUI.
Everything is working fine, but when I try to execute the "magnetic module with FEMM" code I get the following error:

```
  File "C:\Users\snijd\anaconda3\lib\site-packages\pyleecan\Classes\_frozen.py", line 32, in __setattr__
    raise FrozenError(

FrozenError: InputCurrent class has no "N0" property
```

I installed Pyleecan with pip install and have version 1.3.8. I run my code on Spyder with Python version 3.9
Does anyone know how to solve this problem?

With kind regards,

Thomas

BonneelP, 2022-05-16:
Hello, 

I'm currently updating all the tutorials to the latest modifications, the InputCurrent.N0 doesn't exist any more now we have an "OP" object which is more convenient. The new syntax is:
```
from pyleecan.Classes.OPdq import OPdq
simu.input.OP = OPdq(N0=3000, Id_ref=0, Iq_ref=0)
```

Best regards,
Pierre

Theumazz, 2022-05-16:
Dear Pierre,

Thanks a lot for the clarification.
Is there already a source file available (from simulations that others did for example) that I can use for the basic simulations?
It would be useful if I could start from code that is up to date, but I understand that it takes some time to work out the tutorials, so maybe I can use the raw code that you or others used without a complete Jupyter Notebook tutorial.

Thanks in advance,

Thomas Snijders

Theumazz, 2022-05-16:
Dear Pierre, 

Thanks a lot, I will have a look at it! 

With kind regards,

Thomas

BonneelP, 2022-09-06:
Hello
For your information pyleecan 1.4.0 was released and the tutorials are up to date now. 
I'm closing this issue, if you need further help you can reopen it.
Best regards,
Pierre

closed 2022-09-06 by BonneelP