Eomys/pyleecan #555: Github download issue

mandark98, 2022-07-24:
Hi,

I'm a new user of pyleecan still learning to use it. I'm trying to download the tutorials available on github in Eomys/pyleecan. 

When downloading as Zip, some files and folders (Tests, tutorials) are missing in the download. 

I'm new to github, Let me know how can I download the whole thing. 

Thanks in advance for the help!

BonneelP, 2022-07-25:
Hello, 

Thank you for your interest in pyleecan. 

In fact I'm currently updating all the tutorial notebook, I will try to finish by Friday. You can already have a look, most of them are up to date but you may have some surprises (if it is the case please post a message here, I will correct it).  

There are two methods to get the notebook : 
- Fork/Clone the repository https://pyleecan.org/fork.pyleecan.html (that's the best way to get "the whole thing" and being able to modify it)
- Direct download from github https://raw.githubusercontent.com/Eomys/pyleecan/master/Tutorials/tuto_Machine.ipynb (cf https://www.wikihow.com/Download-a-File-from-GitHub)

If you have any question about how pyleecan works don't hesitate to ask questions here.
Best regards,
Pierre

mandark98, 2022-07-30:
Hello Pierre,

Thanks for the reply and encouragement to ask for help.  

The first method suggested by you (Fork/Clone) worked fine. 

I believe the Pyleecan is designed to be operated with jupyter notebook. 

Just to confirm my understanding, When simulating a motor following are the steps to be followed:

1. Open Pyleecan GUI through command window and enter the geometrical details of the motor.
2. Run the tutorials by changing relevant values (File name, Current, time step etc.) to get the required results. 

Kindly suggest if the understanding is in the right direction. 

Thanks in advance,
Mandar Kulkarni.

BonneelP, 2022-09-06:
Hello,

I realize that I didn't answer your last question, sorry for that. I just released pyleecan 1.4.0 and I have updated the website. Now there is a windows installer to run single speed current driven FEMM simulation: https://pyleecan.org/get.pyleecan.html
I have also updated the tutorials and now the page contains a zip with the latest version of the tutorials: https://pyleecan.org/tutorials.html
Yes your 2 steps are correct.

Do you need any further help on the subject or can we close the issue ?
Best regards,
Pierre

mandark98, 2022-09-08:
Hi Pierre, 

Thanks for your reply. 

Yes. We can close this issue. I have been able to open and run some basic simulations through the windows app and through python as well. 

Although, I have some other queries related to loss calculation. I will open another issue for that. 

Thanks Again,
Mandar.

closed 2022-09-08 by mandark98

BonneelP, 2022-09-09:
Great to know :)
For your information, we have an opened PR that rework the losses models and introduce a tutorial #558. It should be the next one I merge but I don't know when I will have time to do that.
Best regards,
Pierre