Eomys/pyleecan #564: [CO] Line Labels

SebGue, 2022-09-21:
Hello,

in #16 there was some discussion about inconvenience regarding line labels. I.e. line labels are defined in line.prop_dict as a key/value pair (e.g. "boundary": "rotor_yoke_side"). 
Now line labels aren't only needed if the lines are boundaries, but also for applying mesh setting, maybe FEA result evaluation etc.
So the key/value pair seem inappropiate. Instead I would suggest to add a label prop. to Line (as in Surface). Further I would rename prop_dict to prop_list that contain all possible properties, e.g. BOUNDARY_PROP_LAB == "Boundary" .

What do you think.

Best regards, Sebastian

BonneelP, 2022-09-22:
Hello,

We used to have line.label but we removed it in #412 (discussion in #344). I don't have time to read the related issue again but I remember that we had long discussions before moving to the prop_dict. This modification was done one year ago, a lot have changed in pyleecan since and we have a better idea of what the prop_dict contains so I agree that we need to take time to improve the current system and write a proper documentation.

If I remember well, we wanted to set too many information in the line label for several purpose so we replaced the label by the prop_dict but maybe a better solution would have been to have label and prop_dict since the label is the most important property to set (and most of the time the only property set). But again different coupling may need different label so it may be tricky to define a generic rule on which label to set, how and why.
Another solution could be to rename BOUNDARY_PROP_LAB  into "LABEL_PROP_LAB" to have a more generic key (or add it as a new key).
I not sure if I understood your point on dict/list and "contain all possible properties" but I don't think that all the lines should have all the property set, since it will create a lot of empty dict or list to handle at each line manipulation. On our side, we have some internal property that we want to set only on few lines, it would be hard to make sure that they are added to all the dict/list (for no purpose other than being there). However, we need to have a better understanding of the possible labels and why they are used for. 

Thank you for opening this discussion, it's an important topic to ease current and future coupling :)

Best regards,
Pierre

SebGue, 2022-09-22:
Hello Pierre,

thank you for your fast answer. I also can't remember the details of this recent discussion.

The point of the list was to store tags (e.g. BOUNDARY_PROP, your_other_internal_prop, ...) for special lines. All other lines will have a line.prop_list = None (or empty list).
But if you got other internal properties, I guess you also use the respective dict value, i.e. line.prop_dict = {internal_prop: prop_value}?! Then a list of properties will be inappropiate.

My other point is that all lines that need some label (e.g. for meshing) need to have the BOUNDARY_PROP_LAB even if they are no boundaries. On the other hand if we use a LABEL_PROP_LAB we could also reintroduce Line.label property.

In general, I think that it is sufficent to have labels that are unique per surface, so every model/coupling can create an overall unique name from surface and line name. Further, I think that 'build_geometry' should include as less information about the models as possible. Otherwise with each new simulation/coupling etc. we need to update all the line labels of all holes/slots etc.

Meanwhile I studied some more code and it seems I'm fine with BOUNDARY_PROP or some other workaround for the moment. 
I will post my findings here when I'm finished my work and we will see.

Best regards, Sebastian