Eomys/pyleecan #569: Problem installing requirements.text

babdwahab, 2022-11-12:
Hi, 
I am new to the pyleecan world and am really excited to use and contribute. I have installed the GUI (method 01). But I want to use it in Jupyter Notebook as it is more flexible. Following the prerequisite on the website (https://pyleecan.org/prerequisite.html#installation) my cmd shell successfully installed all the requirements with some error:

ERROR: Ignored the following versions that require a different python version: 1.21.2 Requires-Python >=3.7,<3.11; 1.21.3 Requires-Python >=3.7,<3.11; 1.21.4 Requires-Python >=3.7,<3.11; 1.21.5 Requires-Python >=3.7,<3.11; 1.21.6 Requires-Python >=3.7,<3.11; 1.6.2 Requires-Python >=3.7,<3.10; 1.6.3 Requires-Python >=3.7,<3.10; 1.7.0 Requires-Python >=3.7,<3.10; 1.7.0rc1 Requires-Python >=3.7,<3.10; 1.7.0rc2 Requires-Python >=3.7,<3.10; 1.7.1 Requires-Python >=3.7,<3.10; 1.7.2 Requires-Python >=3.7,<3.11; 1.7.3 Requires-Python >=3.7,<3.11; 1.8.0 Requires-Python >=3.8,<3.11; 1.8.0rc1 Requires-Python >=3.8,<3.11; 1.8.0rc2 Requires-Python >=3.8,<3.11; 1.8.0rc3 Requires-Python >=3.8,<3.11; 1.8.0rc4 Requires-Python >=3.8,<3.11; 1.8.1 Requires-Python >=3.8,<3.11
ERROR: Could not find a version that satisfies the requirement PySide2 (from versions: none)
ERROR: No matching distribution found for PySide2

Then I saw a post on Git that the requirements are (https://github.com/Eomys/pyleecan/blob/master/requirements.txt). when I do this now I have a different problem:

C:\Users\Admin>pip install -r C:/Users/Admin/Desktop/Pyleecan/requirements.txt
Collecting git+https://gitlab.com/CedMrnl/swat-em.git (from -r C:/Users/Admin/Desktop/Pyleecan/requirements.txt (line 16))
  Cloning https://gitlab.com/CedMrnl/swat-em.git to c:\users\admin\appdata\local\temp\pip-req-build-mh_dgkaw
  ERROR: Error [WinError 2] The system cannot find the file specified while executing command git version
ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?

I also downloaded the Git desktop.

Furthermore, pip install pyleecan gives this error:

C:\Users\Admin>py -m pip install pyleecan
Collecting pyleecan
  Using cached pyleecan-1.4.0.tar.gz (11.5 MB)
  Preparing metadata (setup.py) ... done
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
pyleecan depends on swat-em@ git+https://gitlab.com/CedMrnl/swat-em.git

I am not sure what should I do. I have these versions:

Python: 3.11.0
Pip: 22.3.1

Thank you for your help.

babdwahab, 2022-11-13:
Now I am getting this error:

C:\Users\Admin>pip install pyleecan
Collecting pyleecan
  Using cached pyleecan-1.4.0.tar.gz (11.5 MB)
  Preparing metadata (setup.py) ... done
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
pyleecan depends on swat-em@ git+https://gitlab.com/CedMrnl/swat-em.git

RaphaelPile, 2022-11-15:
Hi @babdwahab, 

For the first part of your problem, the cause of the failing package installations is quite obvious to me: you are using Python 3.11 while most of these packages need a Python version between 3.7 and 3.8.

If I'm not mistaken, we recommended to use Pyleecan with Python 3.8. Try that before looking into any other problem. 

Best regards,

Raphaël

babdwahab, 2022-11-16:
Hello,
Thanks for your feedback. I have done as you said. First I run:
C:\Users\Admin>pip install git+https://gitlab.com/CedMrnl/swat-em.git 

which does install swat-em successfully. Later I run:

C:\Users\Admin>pip install pyleecan
Collecting pyleecan
  Using cached pyleecan-1.4.0.tar.gz (11.5 MB)
  Preparing metadata (setup.py) ... done
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
pyleecan depends on swat-em@ git+https://gitlab.com/CedMrnl/swat-em.git

It still gives me the same swat error. Please let me know what can I do. Thank you

I am using python 3.8.10.

babdwahab, 2022-11-18:
Hurray!!! It was successfully installed without any problem.  Thank you for your support...

BonneelP, 2022-11-18:
Great, welcome to pyleecan :)  Tutorials are available here with an archive to get the latest notebook (updated around September): https://pyleecan.org/tutorials.html

I'm closing this issue then, feel free to open a new one if you need further assistance with pyleecan. 

Best regards,
Pierre

closed 2022-11-18 by BonneelP