Eomys/pyleecan #571: Custom rotor and magnet shape

babdwahab, 2022-11-22:
Dear all, I am trying to model a machine which has a rotor lamination as shown below. The orange is the rectangular magnet. The shape is nothing unique but when I am defining my machine in GUI and I could not get this shape not even close (see attached pic). I was wondering how you folks deal with it and if there is a document which walks through setting custom slots and rotor magnets using scripts I would appreciate if you let me know.  Looking forward to your response. Thank you very much.

[image]

[image]

BonneelP, 2022-11-22:
Hello,

Thank you for your interest in pyleecan. 

I think that you may have selected the wrong machine kind. This step and schematics is available only for SPMSM (Surface Permanent Magnet Synchronous Machine) and the screenshot of your machine seems to be an IPMSM (Interior Permanent Magnet Synchronous Machine). You can change the machine type in the first step of the GUI but you will have to enter again all the parameters of your machine.

Then regarding your particular geometry I see 3 options:
- First you can use one of the following schematics (HoleM52 and HoleM52R) but it's not exactly what you are looking for:
[image]
[image]
- If not close enough you can use the DXF import to draw exactly your topology but you would need a DXF of your machine and you won't be able to edit your topology (no parameter sweep on the magnet size for instance).
- We can guide you to define a schematics to add your topology into pyleecan, then you will have exactly your topology and the possibility to edit its parameters. 

Best regards,
Pierre

SebGue, 2022-11-22:
Hello babdwahab,

for the rotor you try to model, you should setup a interior magnet machine (MachineIPMSM) with HoleM50 rotor.

[image] 

Best regards, Sebastian

babdwahab, 2022-11-22:
Hello Sebastian and Pierre,
Thank you for your suggestion. i did find the rotor shape I was looking for in IPMSM motor. I got it pretty close by using Hole52. I work on custom designs most of the time and I am sure there will be point where it will be really difficult to use the existing slot and rotor holes available. This is one of the reasons I moved to Pyleecan to be able to generate my own parts library. DXF files defeats the purpose of freely optimizing the design. I am very new to Pyleecan and don't really know how I can make those custom parts using schematics as Pierre mentioned. Your help will be really appreciated to get one of the topologies build and I can carry on from there. Are you going to share with me a sample or documentation which builds a custom part or are we going to work through a part that I have? Please let me know. Thank you very much.

Bests,
Abdul 

[image]

BonneelP, 2022-11-22:
Hello, 

Adding a new Hole/Schematics in pyleecan is not that hard once you know how pyleecan works. The trickiest part is to compute the coordinates of all the points according to your schematics. The full process is:
- Get the code from pyleecan to contribute to it (Method 3 of https://pyleecan.org/get.pyleecan.html)
- Draw "By hand" your schematics: What is the shape of the hole you want to draw ? What are the parameters and their exact definition ?
- Create a csv file (by copy/paste/adapt existing ones from pyleecan\Generator\ClassesRef\Slot) to define the new class that will represent your Hole
- Create the methods of your Hole, start by _comp_point_coordinate then build_geometry (all the other should be optional - to check)
- Add some Tests to make sure that your methods work and always will by copy/paste/adapt 
- Add the plot_schematics method to create a clean standard image as the other Hole
- Add the GUI related widget by copy/paste/adapt 
- Add the Tests on the GUI widget by copy/paste/adapt 

Technically, once you have define the two methods _comp_point_coordinate and build_geometry you should be able to use your hole but it is better to do the full process to interact with your schematics inside the GUI. 
The general process is further detailed in this old (maybe outdated) webinar: https://pyleecan.org/webinar_3.html and this tutorial https://pyleecan.org/tuto.add.slot.html

If you are interesting in adding your topology inside pyleecan, I can guide you step by step (and write some part of the code like the Tests for instance). If needed I can also take the opportunity to update the website to answer your questions.

Best regards,
Pierre

babdwahab, 2022-11-23:
Hi Pierre,
I just watched the webinar you mentioned and got some valuable insights on what I need to do. Thanks for that. I will make a list of things that need to be accomplished (slot geometry) and let you know if I need help. Certainly, I will :)

Bests,
Abdul