Eomys/pyleecan #578: Issues in installing Pyleecan v1.4.1

crowdnidhi-design, 2022-12-11:
Dea Sir/Madam,

Iam trying to install Pyleecan v1.4.1 ( latest version) but my antivirus software ( Norton) is detecting too many viruses.
Can you please help or provide us a safe installable latest exe.

crowdnidhi-design, 2022-12-16:
Sure let me try. Further, let me uninstall Norton and try again.

crowdnidhi-design, 2022-12-17:
Norton is removed hence I was able to install it and works ok.
Can you please share with me detailed steps to install the full package including python on windows 11?

BonneelP, 2022-12-19:
Hello,

The full procedure is detailed on this page: https://pyleecan.org/get.pyleecan.html
Is you have any questions, it could help us improve this page.

Best regards,
Pierre