Eomys/pyleecan #580: Installing issue - windows 11

crowdnidhi-design, 2022-12-17:
When I try to install Pyleecan I get below error

ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
pyleecan depends on swat-em@ git+https://gitlab.com/CedMrnl/swat-em.git

I have Python 3.11.1 running on my machine.

Can you please help me fix the issue

BonneelP, 2022-12-19:
Hello,

Normally this issue was solved with the release 1.4.1 cf #569

Maybe the issue is linked to 3.11, we are currently using python 3.8 and we need to investigate the package compatibility for other versions. Is 3.8 an option for you ? If not you can try to run:
`pip install git+https://gitlab.com/CedMrnl/swat-em.git`
then run 
`pip install pyleecan`

Best regards,
Pierre

crowdnidhi-design, 2022-12-19:
Well, I completed the installation using python 3.8.x. 
Now when I am trying to do an example of an IPM motor I am getting swat-em not available and fails during the winding configuration step. What should I try next

crowdnidhi-design, 2022-12-20:
Well I resolved the error now I am getting 
[10:25:58] Error while running simulation:
unsupported operand type(s) for -: 'float' and 'NoneType'
[10:26:00] Error while running simulation:
unsupported operand type(s) for -: 'float' and 'NoneType'
[10:26:00] Error while saving results:
local variable 'out' referenced before assignment
[10:26:03] Error while saving results:

BonneelP, 2022-12-20:
Hello,

Can you share the script/simulation/machine that you are trying to run ? 

Best regards,
Pierre

crowdnidhi-design, 2022-12-20:
Dear Pierre,

I have resolved the issue. Well, I guess I had entered the wrong data hence it was throwing an error.

 Let me try one more time all features/functionality.  

Further,

1. Just curious do you have MTPA, FOC, or DTC options/libraries /Algo available as a functionality/feature ?
2. Can we integrate Pyleecan with Scilab/XCOS?.